ElettraSciComp/witmotion_IMU_ros #2: Provide installation guidance in Readme.md

tobikellner, 2022-06-13:
Please provide some installation guidance within the readme.

Had to figure out the hard way which required packages where missing. Could look like:

## Installation
1. clone the repositiory into your catkin_ws/src folder using --recursive to also clone the linked witmotion-uart-qt module: 
```sh 
cd catkin_ws/src
git clone --recursive https://github.com/ElettraSciComp/witmotion_IMU_ros.git
```
if the recursive clone fails clone the uart-qt repository manually into the witmotion-uart-qt folder

2. install qt5serialPort with
```sh
sudo apt-get install libqt5serialport5-dev
```

rlabs-oss, 2022-06-23:
depdance time, please write instructions that work as tobikellner above requests - this package doesnt work. 
/src/witmotion_IMU_ros/witmotion-uart-qt

  does not contain a CMakeLists.txt file.

package lacks <depend>std_srvs</depend>

please provide a multistep install for the qt / uart package, then the ros node package.

twdragon, 2022-06-24:
@rlabs-oss you need just to make a recursive clone.

I will provide tge installation guide soon. Thank you for the feedback!

rlabs-oss, 2022-06-24:
> @rlabs-oss you need just to make a recursive clone.
> 
> I will provide tge installation guide soon. Thank you for the feedback!

As mentioned that didn't work so I manually cloned the required parts. I'm happy to report it is working now, I also contacted witmotion for support on configuring the 901c. 
Many thanks.

git clone --recursive https://github.com/ElettraSciComp/witmotion_IMU_ros
Cloning into 'witmotion_IMU_ros'...
remote: Enumerating objects: 67, done.
remote: Counting objects: 100% (67/67), done.
remote: Compressing objects: 100% (35/35), done.
remote: Total 67 (delta 28), reused 59 (delta 23), pack-reused 0
Unpacking objects: 100% (67/67), 17.92 KiB | 1.19 MiB/s, done.
Submodule 'witmotion-uart-qt' (git@github.com:ElettraSciComp/witmotion_IMU_QT.git) registered for path 'witmotion-uart-qt'
Cloning into '/home/lab/test/witmotion_IMU_ros/witmotion-uart-qt'...
The authenticity of host 'github.com (140.82.121.3)' can't be established.
ECDSA key fingerprint is SHA256:p2QAMXNIC1TJYWeIOttrVc98/R1BUFWu3/LiyKgUfQM.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'github.com,140.82.121.3' (ECDSA) to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:ElettraSciComp/witmotion_IMU_QT.git' into submodule path '/home/lab/test/witmotion_IMU_ros/witmotion-uart-qt' failed
Failed to clone 'witmotion-uart-qt'. Retry scheduled
Cloning into '/home/lab/test/witmotion_IMU_ros/witmotion-uart-qt'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:ElettraSciComp/witmotion_IMU_QT.git' into submodule path '/home/lab/test/witmotion_IMU_ros/witmotion-uart-qt' failed
Failed to clone 'witmotion-uart-qt' a second time, aborting

twdragon, 2022-06-24:
@rlabs-oss it will be also really great if you would provide any documentation for 901C sensor to examine for differences in command formulation and tests for the library!

rlabs-oss, 2022-06-24:
witmotion sent
"Please check the relevant information about WT901C-232.

https://drive.google.com/file/d/1WPgvIaYERJsWvHeQy4A3UGKF7UWm9aJ8/view?usp=sharing"
----

For the intended application Im not sure if the EKF is causing problems since the Lidar when used with FASTER_LIO assumes raw readings and it computes its own pose using iterative EKF.  Ive set the filter to 286hz and output rate to 200hz, with baud at 230400. The witmotion 901 is generating the orientation in addition to the acceleration and velocity. The Lidar (in this example a Livox Avia) sets the orientation quarternions to 0 and angular_velocity and lin acceleation as the witmotion IMU, so that looks "ok" - however in practice using the Livox Mid 40 (a lidar without IMU) the scan is invalid. Filter above 200hz according to the witmotion manual turns off filtering  - but I need to debug this more. Do you know of any ros tools that will generate a plot of the IMU data so I can compare them?  

Couple of examples below from the Avia and 901C

The Livox Avia IMU example:-
header: 
  seq: 5299
  stamp: 
    secs: 141
    nsecs: 900395480
  frame_id: "livox_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.04248914122581482
  y: 0.004896665457636118
  z: -0.1790435016155243
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.07477080821990967
  y: 0.03358038514852524
  z: 1.0024305582046509
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---

Witmotion 901C
header: 
  seq: 28705
  stamp: 
    secs: 1656085951
    nsecs: 110546361
  frame_id: "imu"
orientation: 
  x: 0.012426594989697597
  y: -0.01768112063882205
  z: 0.9946913368802636
  w: 0.10060865790413762
orientation_covariance: [0.021299999207258224, 0.0, 0.0, 0.0, 0.042899999767541885, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.02130528911948204
  y: -0.19600865244865417
  z: -0.2780340015888214
angular_velocity_covariance: [0.007499999832361937, 0.0, 0.0, 0.0, 0.0038999998942017555, 0.0, 0.0, 0.0, 0.0034000000450760126]
linear_acceleration: 
  x: -1.0202783346176147
  y: -2.2417383193969727
  z: 9.44595718383789
linear_acceleration_covariance: [0.007499999832361937, 0.0, 0.0, 0.0, 0.0038999998942017555, 0.0, 0.0, 0.0, 0.0034000000450760126]

twdragon, 2022-06-30:
@rlabs-oss thank you! Added to the wiki. Working on the guide

twdragon, 2022-06-30:
@rlabs-oss Can you please also open a separate issue for this discussion?

closed 2022-06-30 by twdragon

rlabs-oss, 2022-06-30:
Perhaps better to mention that the ros imu_plugin will visualize the state of the IMU in rviz to the wiki. 
I can provide a .bag file and example how to start roscore, record the IMU data and and play it back (from the .bag) file- which I found was sufficient to show basic info. - Ill start that in a new discussion once its ready. 
Thank you.

twdragon, 2022-06-30:
@rlabs-oss if you have also some graphics or a short screencast with IMU state visualization, could you please add the information about visualization (with the link to the plugin) to ROS wiki? I am now overloaded a bit.

commit 05c5a6c7 (2022-06-30): Readme fix, closes #2 after adding installation guile