ElettraSciComp/witmotion_IMU_ros #6: clone method still fails

rlabs-oss, 2022-07-06:
$ git clone --recursive https://github.com/ElettraSciComp/witmotion_IMU_ros.git src/witmotion_ros
Cloning into 'src/witmotion_ros'...
remote: Enumerating objects: 98, done.
remote: Counting objects: 100% (98/98), done.
remote: Compressing objects: 100% (54/54), done.
remote: Total 98 (delta 46), reused 82 (delta 33), pack-reused 0
Unpacking objects: 100% (98/98), 22.91 KiB | 350.00 KiB/s, done.
Submodule 'witmotion-uart-qt' (git@github.com:ElettraSciComp/witmotion_IMU_QT.git) registered for path 'witmotion-uart-qt'
Cloning into '/home/lab/external-source/src/witmotion_ros/witmotion-uart-qt'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:ElettraSciComp/witmotion_IMU_QT.git' into submodule path '/home/lab/external-source/src/witmotion_ros/witmotion-uart-qt' failed
Failed to clone 'witmotion-uart-qt'. Retry scheduled
Cloning into '/home/lab/external-source/src/witmotion_ros/witmotion-uart-qt'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:ElettraSciComp/witmotion_IMU_QT.git' into submodule path '/home/lab/external-source/src/witmotion_ros/witmotion-uart-qt' failed
Failed to clone 'witmotion-uart-qt' a second time, aborting

twdragon, 2022-07-08:
I tried to reproduce this error without any success. It seems that your Git installation has the wrong SSH keychain setup. You need just to use HTTPS clone protocol or to set up SSH key in your Github account

twdragon, 2022-07-08:
I will add HTTPS to README.

Implementation of sensor settings control is WIP now, hope to do it soon. @rlabs-oss please do not mix the thread handling different issues.