ElettraSciComp/witmotion_IMU_ros #7: any plan to support ros2?

kyuhyong, 2022-07-08:
It would be great if wit imu support ros2 as well. 
Any time soon to release the driver?

twdragon, 2022-07-08:
Now it is sadly not a priority task for me. To help with this, you also can participate by trying to compile  the forked branch of the repository on ROS2 and reporting the errors here or in the dedicated issues, so we would easily discover incompatibilities

fllay, 2022-10-14:
I have tried by the submodule create a problem.

```
pi@pi-desktop:~/amr_ws$ colcon build --packages-select witmotion_ros --symlink-install
Starting >>> witmotion_ros
--- stderr: witmotion_ros                         
CMake Error at CMakeLists.txt:12 (message):
  witmotion-uart-qt directory does not contain submodule data.  Please update
  submodules to compile the underlying library in-place for ROS integration

---
Failed   <<< witmotion_ros [0.25s, exited with code 1]

Summary: 0 packages finished [1.20s]
  1 package failed: witmotion_ros
  1 package had stderr output: witmotion_ros
```

The code was generated by using 

[](https://github.com/awslabs/ros2-migration-tools)

Is it possible to remove submodule? Just include everything in one repo. Thank you

twdragon, 2022-10-14:
@fllay did you try:
```sh
git submodule update --init --recursive
```
before compiling? The submodule is necessary to separate versions of the node and the library.

fllay, 2022-10-15:
I tried
```
git submodule update --init --recursive
```
 It can be build now but I still get error 
```
Starting >>> witmotion_ros
--- stderr: witmotion_ros                         
CMake Error at /usr/lib/aarch64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by "Qt5SerialPort"
  with any of the following names:

    Qt5SerialPortConfig.cmake
    qt5serialport-config.cmake

  Add the installation prefix of "Qt5SerialPort" to CMAKE_PREFIX_PATH or set
  "Qt5SerialPort_DIR" to a directory containing one of the above files.  If
  "Qt5SerialPort" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  witmotion-uart-qt/CMakeLists.txt:30 (find_package)

---
Failed   <<< witmotion_ros [1.74s, exited with code 1]

Summary: 0 packages finished [3.13s]
  1 package failed: witmotion_ros
  1 package had stderr output: witmotion_ros
```

Then I install  `sudo apt-get install libqt5serialport5-dev`. I still git this error

```
--- stderr: witmotion_ros                         
CMake Error at CMakeLists.txt:54 (find_package):
  By not providing "FindSerialPort.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "SerialPort", but CMake did not find one.

  Could not find a package configuration file provided by "SerialPort" with
  any of the following names:

    SerialPortConfig.cmake
    serialport-config.cmake

  Add the installation prefix of "SerialPort" to CMAKE_PREFIX_PATH or set
  "SerialPort_DIR" to a directory containing one of the above files.  If
  "SerialPort" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< witmotion_ros [5.78s, exited with code 1]

Summary: 0 packages finished [6.73s]
  1 package failed: witmotion_ros
  1 package had stderr output: witmotion_ros
```

Is this a `roserial` package in ros1?

kyuhyong, 2022-10-16:
I actually implemented the driver for ROS 2 based on https://github.com/Rophier/WIT-IMU here.
Here is my implementation of the same sensor. 
https://github.com/kyuhyong/WIT-IMU

fllay, 2022-10-17:
@twdragon Thank you. Now I can fix that issue and I am re-writing ros2 pub but I don't know it will be ok or not.

@kyuhyong  Thank you I will give it a try. I have only WT61C TTL and it does not work out of the box. I ordered 
WT901C and will see. 

BTW WT61C TTL works with  `[witmotion_IMU_ros](https://github.com/ElettraSciComp/witmotion_IMU_ros)
Public` without any modification. Just edit the config file and It is very stable.

twdragon, 2022-10-17:
@fllay looking forward to see the PR from you. Really!

fllay, 2022-10-21:
I'm willing to PR but I don't know how to do PR or how PR works. I edited `witmotion_ros.cpp` and `witmotion_ros.h` to use ros2 service and publisher. It seems to be compiled OK but I have trouble with `witmotion_ros_node.cpp`. It complained about link error. I just used the following code in `witmotion_ros_node.cpp`

```
#include "witmotion_ros.h"

#include <signal.h>
#include <stdlib.h>
#include <stdio.h>
#include <unistd.h>

void handle_shutdown(int s)
{
//    RCLCPP_INFO(->get_logger(),"Shutting down node...");
//    rclcpp::shutdown();
    QCoreApplication::exit(0);
}

int main(int argc, char * argv[])
{ 
    ROSWitmotionSensorController& controller = ROSWitmotionSensorController::Instance();
    controller.Start();
}
```

and the gave error message 

```
/usr/bin/ld: CMakeFiles/witmotion_ros_node.dir/src/witmotion_ros_node.cpp.o: in function `main':
witmotion_ros_node.cpp:(.text+0x18c): undefined reference to `ROSWitmotionSensorController::Instance()'
/usr/bin/ld: witmotion_ros_node.cpp:(.text+0x198): undefined reference to `ROSWitmotionSensorController::Start()'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/witmotion_ros_node.dir/build.make:4671: witmotion_ros_node] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:158: CMakeFiles/witmotion_ros_node.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< witmotion [1min 1s, exited with code 2]
                                     
Summary: 0 packages finished [1min 2s]
  1 package failed: witmotion
  1 package had stderr output: witmotion
```

I  don't know how to fix this.  I don't even start ros2 node yet. Any idea?

twdragon, 2022-10-21:
@fllay did you delete `Instance()` function? If not, did you link the underlying library properly via `target_link_libraries`?

twdragon, 2022-10-21:
The pull request will request us to allow adding your code as a particular branch to our repository. After the colleagues approve, I will test and hold your contribution as part of the project/

twdragon, 2022-10-23:
@fllay I examined your pull request. You cancelled compilation of an underlying library and the ROS library, so there is nothing to link to the node. You should review `CMakeLists.txt` to allow the libraries to be compiled.

twdragon, 2022-11-14:
Development of the ROS2 version moved to `ros2` branch.

closed 2022-11-14 by twdragon