ElettraSciComp/witmotion_IMU_ros #9: Не запускается

tohax, 2022-07-25:
[witmotion_imu-2] process has died [pid 16036, exit code -11, cmd /home/tohax/ws_core/devel/lib/witmotion_ros/witmotion_ros_node __name:=witmotion_imu __log:=/home/tohax/.ros/log/7c5c6e64-0c13-11ed-b4bf-00155d39e248/witmotion_imu-2.log].
log file: /home/tohax/.ros/log/7c5c6e64-0c13-11ed-b4bf-00155d39e248/witmotion_imu-2*.log

В чем может быть причина?

twdragon, 2022-07-25:
Если сразу умирает, скорее всего, нет прав на доступ к порту или неправильное имя порта в конфигурации. Пришлите полный лог, если есть.

tohax, 2022-07-26:
Итак. Я работаю в WSL. Установлена Ubuntu 20.04. Ros Noetic.
 Модуль WT901CTTL прокинут в Ubuntu из Windows 11 c помощью usbip.
 Другие драйверы под ROS для этого модуля работают, но нет возможности установить частоту отдачи данных 200hz.
 Решил протестировать ваш драйвер для ROS. В настройках драйвера (config.yaml) ставлю port: ttyUSB0 и baud_rate: 115200.
 После catkin_make делаю source ~/ws_core/devel/setup.sh
 Сделал sudo chmod 777 /dev/ttyUSB0.
 Не помогает.
 
 
 roslaunch witmotion_ros witmotion.launch
... logging to /home/tohax/.ros/log/bccccd92-0ca2-11ed-86ad-00155d57dcbf/roslaunch-MAKUSHIN-2086.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://MAKUSHIN:34635/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /witmotion_imu/altimeter_publisher/addition: 0.0
 * /witmotion_imu/altimeter_publisher/coefficient: 1.0
 * /witmotion_imu/altimeter_publisher/enabled: True
 * /witmotion_imu/altimeter_publisher/topic_name: /altitude
 * /witmotion_imu/barometer_publisher/addition: 0.0
 * /witmotion_imu/barometer_publisher/coefficient: 1.0
 * /witmotion_imu/barometer_publisher/enabled: True
 * /witmotion_imu/barometer_publisher/frame_id: base_link
 * /witmotion_imu/barometer_publisher/topic_name: /barometer
 * /witmotion_imu/barometer_publisher/variance: 0.001
 * /witmotion_imu/baud_rate: 115200
 * /witmotion_imu/gps_publisher/enabled: False
 * /witmotion_imu/gps_publisher/ground_speed_topic_name: /gps_ground_speed
 * /witmotion_imu/gps_publisher/navsat_altitude_topic_name: /gps_altitude
 * /witmotion_imu/gps_publisher/navsat_fix_frame_id: world
 * /witmotion_imu/gps_publisher/navsat_fix_topic_name: /gps
 * /witmotion_imu/gps_publisher/navsat_satellites_topic_name: /gps_satellites
 * /witmotion_imu/gps_publisher/navsat_variance_topic_name: /gps_variance
 * /witmotion_imu/imu_publisher/frame_id: imu
 * /witmotion_imu/imu_publisher/measurements/acceleration/covariance: [0.0075, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/acceleration/enabled: True
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/covariance: [0.0075, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/enabled: True
 * /witmotion_imu/imu_publisher/measurements/orientation/covariance: [0.0213, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/orientation/enabled: True
 * /witmotion_imu/imu_publisher/topic_name: /imu
 * /witmotion_imu/magnetometer_publisher/addition: 0.0
 * /witmotion_imu/magnetometer_publisher/coefficient: 1.0
 * /witmotion_imu/magnetometer_publisher/covariance: [0.01, 0, 0, 0, 0...
 * /witmotion_imu/magnetometer_publisher/enabled: True
 * /witmotion_imu/magnetometer_publisher/frame_id: imu
 * /witmotion_imu/magnetometer_publisher/topic_name: /magnetometer
 * /witmotion_imu/orientation_publisher/enabled: False
 * /witmotion_imu/orientation_publisher/topic_name: /orientation
 * /witmotion_imu/polling_interval: 30
 * /witmotion_imu/port: ttyUSB0
 * /witmotion_imu/restart_service_name: /restart_imu
 * /witmotion_imu/rtc_publisher/enabled: False
 * /witmotion_imu/rtc_publisher/topic_name: /witmotion_clock
 * /witmotion_imu/temperature_publisher/addition: 0.0
 * /witmotion_imu/temperature_publisher/coefficient: 1.0
 * /witmotion_imu/temperature_publisher/enabled: True
 * /witmotion_imu/temperature_publisher/frame_id: base_link
 * /witmotion_imu/temperature_publisher/from_message: magnetometer
 * /witmotion_imu/temperature_publisher/topic_name: /temperature
 * /witmotion_imu/temperature_publisher/variance: 0.1

NODES
  /
    witmotion_imu (witmotion_ros/witmotion_ros_node)

auto-starting new master
process[master]: started with pid [2108]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bccccd92-0ca2-11ed-86ad-00155d57dcbf
process[rosout-1]: started with pid [2132]
started core service [/rosout]
process[witmotion_imu-2]: started with pid [2139]
Opening device "ttyUSB0" at 115200 baud
Instantiating timer at 30 ms
[witmotion_imu-2] process has died [pid 2139, exit code -11, cmd /home/tohax/ws_core/devel/lib/witmotion_ros/witmotion_ros_node __name:=witmotion_imu __log:=/home/tohax/.ros/log/bccccd92-0ca2-11ed-86ad-00155d57dcbf/witmotion_imu-2.log].
log file: [/home/tohax/.ros/log/bccccd92-0ca2-11ed-86ad-00155d57dcbf/witmotion_imu-2*.log]

twdragon, 2022-07-26:
@tohax для 200 Гц нужно:
- Выяснить, поддерживает ли сенсор такую частоту отдачи (документация Witmotion часто недостоверна: указано до 200 Гц, на практике поддержка есть только до 100 Гц).
- Передать на сенсор явным образом командные пакеты, устанавливающие baudrate 115200 и частоту 200 Гц, потом сохранить настройки. 
- Учесть, что частота опроса может быть немного ниже частоты отдачи, так как частота отдачи сенсора - это частота выдачи пакетов, а системе требуется время на расчет контрольных сумм и сборку сообщений IMU для ROS.

Попробуйте поэкспериментировать с параметрами, чтобы определить, какой точно режим на данный момент установлен на вашем датчике. По умолчанию на WT901C порт устанавливается в 9600 бод с частотой выдачи пакетов 10 Гц.

На данный момент приложение-контроллер, поддерживающее выдачу команд на сенсоры семейства WT901C находится в стадии разработки (надеюсь добавить его через несколько дней). Можно воспользоваться кодами команд из заголовка библиотеки низкого уровня [`types.h`](https://github.com/ElettraSciComp/witmotion_IMU_QT/blob/main/include/witmotion/types.h), чтобы написать простейший установщик параметров.

tohax, 2022-07-26:
Изначально, в программе под Windows, которая шла с модулем WT901C, я установил значения 115200 и 200hz. При использовании других драйверов для ROS модуль подключается и передает данные. Отложим частоту 200hz. Хочется заставить мой WT901C работать с вашим драйвером хоть с какой-нибудь частотой. Что еще может быть причиной ошибки, описанной ранее? Спасибо.

twdragon, 2022-07-26:
Код -11 это Segmentation Fault. 

Попробуйте собрать отдельно библиотеку низкого уровня и запустить из нее приложение `message-enumerator` с ключами `--baudrate 115200` и `--poll 20`. Тогда сразу поймем, что не так с портом. Возможно, есть проблемы с установкой Qt, возможно, неправильный линк из Windows.

twdragon, 2022-07-26:
@tohax только что тестировал мой бескорпусный WT901, стабильно работает на 100 Гц, 115200 бод, 10 мс таймер, но в чистом Linux

twdragon, 2022-07-26:
@tohax еще можно попробовать в каталог `witmotion-uart-qt` сделать клон библиотеки из ветки `main`, на случай, если в модуле неправильный коммит прописан

tohax, 2022-07-26:
"Попробуйте собрать отдельно библиотеку низкого уровня и запустить из нее приложение message-enumerator с ключами --baudrate 115200 и --poll 20. Тогда сразу поймем, что не так с портом. Возможно, есть проблемы с установкой Qt, возможно, неправильный линк из Windows."

Как это сделать, я, к сожалению, не знаю. Подскажите команду, пожалуйста.

Клонировал ветку main, так как "git clone -recursive" не клонировал witmotion-uart-qt.

Еще, во время сборки драйвера, идут ошибки по Qt::endl, Qt::hex, Qt:dec. Пришлось везде убрать "Qt::". У меня Версия Qt 5.12. Может в этом причина?

twdragon, 2022-07-26:
@tohax тогда все, скорее всего, из-за того, что библиотека не клонируется в нужный каталог сборки. Раз ее нет, подключать нечего, и возникает ошибка вызова по адресу функции. Попробуйте повторить клон, либо командой 
```
git clone --recursive https://github.com/ElettraSciComp/witmotion_IMU_ros.git witmotion_ros
```
либо такими командами:
```
git clone https://github.com/ElettraSciComp/witmotion_IMU_ros.git witmotion_ros
cd witmotion_ros
git submodule update --init --recursive
```
Тогда клон подхватит нужные модули. 

В любом случае, каталог `witmotion-uart-qt` должен содержать код из репозитория https://github.com/ElettraSciComp/witmotion_IMU_QT

На Qt 5.12.1 я тестировал, с пространствами имен проблем нет. Возможно, ваша установка ROS перекрыла пути к библиотекам Qt разных версий в Windows Linux Subsystem. Для чистоты эксперимента, я попробовал бы собрать отдельно ROS Workspace с библиотекой в контейнере Docker, либо виртуальной машине с Ubuntu, на которой установлен ROS.

tohax, 2022-07-26:
Вручную клонировал файлы из witmotion_IMU_QT в witmotion-uart-qt. Все файлы на своих местах)
[image]

Установил Virtualbox, Ubuntu 20.04. Опять удалил "Qt::" из всех файлов, которые появляются с ошибками во время catkin_make. Версия Qt 5.12.8.
Та же ошибка - exit code -11. Что же я делаю не так? Спасибо огромное за помощь

twdragon, 2022-07-26:
Попробуйте в файле конфигурации отключить публикацию всех сообщений, кроме IMU. Можете скопировать сюда ваш конфиг? Посмотрим соответствие публикуемых данных настройкам.

Я проверил сейчас подключение библиотеки в ROS, все работает. Бампнул коммит в репозиторий, можно повторно сделать клон. Еще посоветую после реклона ДО запуска `catkin_make` удалить каталоги `build` и `devel`, чтобы по инерции CMake старые бинарники не хватал из кэша

twdragon, 2022-07-26:
@tohax есть вероятность, что все так, просто в Ubuntu сломалась установка Qt. У вас `libqt5serialport5-dev` установлен?

twdragon, 2022-07-26:
Библиотеку можно проверить так: в отдельном каталоге вне ROS:

```
git clone https://github.com/ElettraSciComp/witmotion_IMU_QT.git witmotion
cd witmotion
mkdir build && cd build
cmake ..
make
./message-enumerator --baudrate 115200 --poll 30
```

После этого ждем, будет ли мигать светодиод адаптера UART, потом жмем `Ctrl+C` и анализируем отчет о сообщениях, или сообщения об ошибках.

tohax, 2022-07-26:
./message-enumerator --device ttyUSB0 --baudrate 115200 --poll 10
Press Ctrl+C to stop enumeration and see the report
Opening device "ttyUSB0" at 115200 baud
Instantiating timer at 10 ms
ERROR: No data acquired during last 3 iterations, please check the baudrate!
Closing TTL connection

WITMOTION UART MESSAGE ENUMERATOR BY TWDRAGON

Acquired at Tuesday, 26 July 2022 18:03:31 +05

ID      Qty     Description

0x50    1       Real Time Clock
0x51    1       Accelerations
0x52    1       Angular velocities

        Unknown IDs: 0 [  ]
Total messages: 3

О5 пришлось удалять Qt::
При каждом запуске разное количество сообщений (Total messages) Может быть вообще 0, а может и 4. Что это значит?

twdragon, 2022-07-26:
Это значит, что реальная частота выдачи данных на датчике меньше 100 Гц (на практике, там скорее всего, 10-20 Гц). Программа реагирует, если таймер сработал вхолостую более 3 раз. Попробуйте подобрать значения таймера (`--poll`) и скорости передачи (`--baudrate`) для стабильной работы,  чтобы программа смогла принять хотя бы сотню сообщений.

С `Qt::` все плохо: версии до и после 5.14 официально несовместимы на уровне имен. Подумаю, как с этим быть.

twdragon, 2022-07-26:
`Qt::endl` победил, версию в модуль пропишу после тестирования.

tohax, 2022-07-26:
Это значит, что реальная частота выдачи данных на датчике меньше 100 Гц (на практике, там скорее всего, 10-20 Гц).
Вот это меня и волновало. 
Мне нужен датчик, который отдается данные 200гц и больше. Не подскажете такой?

twdragon, 2022-07-26:
Вот этот вроде, выдавал
https://www.phidgets.com/?tier=3&catid=10&pcid=8&prodid=1025

Если будет успешный тест WT901 на 200 Гц (правда, это явно предел для китайцев), выложу его сюда

tohax, 2022-07-26:
Спасибо большое!

twdragon, 2022-07-27:
@tohax в дополнение: сейчас пытался заставить работать WT901 с помощью команд из официальной документации(!)

Результат нулевой: датчик стоит на 9600 бод, 10 Гц. На команды конфигурации не реагирует. Сейчас написал китайцам, пусть разбираются

tohax, 2022-07-27:
Да, хорошо. Ждем ответа. Скачал последний commit. Qt::hex ругается.

twdragon, 2022-07-27:
> Да, хорошо. Ждем ответа. Скачал последний commit. Qt::hex ругается.

Победим в следующем коммите

twdragon, 2022-07-27:
@tohax обсуждение переносим [сюда](https://github.com/ElettraSciComp/witmotion_IMU_QT/issues/2), так как проблемы с датчиками вскрыли необходимость аудита фирменной документации Witmotion.

twdragon, 2022-07-28:
@tohax протестил WT901, настроенный с помощью приложения-контроллера на 115200 бод, 200 Гц. Удалось на 7-10 мс периоде опроса вытянуть 180 Гц. Больше, видимо, не получится - сенсор не успевает выдавать нужное количество пакетов до следующего запроса ядра к порту. Как я и говорил, фактическое временное разрешение этого датчика - не более 175 Гц, максимальное - оценочно до 190, больше даже производитель не заявляет.

tohax, 2022-07-29:
А если поднять до 230400-921600 бод? Официальное приложение позволяет выставить. Или это не повлияет?

twdragon, 2022-07-29:
> А если поднять до 230400-921600 бод? Официальное приложение позволяет выставить. Или это не повлияет?

Думаю, влияния не будет от слова "совсем".

$200 Hz \cdot 100 bits/block = 20000 baud$, то есть, там избыточности по скорости модуляции выше крыши 38400 бод перекроет все,  что можно. Проблема в том, что датчик выдает с частотой 200 Гц сами пакеты с измерениями, то есть последовательно: ускорения - скорости - углы - магнит, а на практике на само формулирование пакетов тратится время и если считать блоками, необходимыми для IMU, получается, что только **каждый пятый пакет** начинает новый блок. Тогда между последовательными съемами в одном измерении всех пакетов,  пригодных для сборки сообщения IMU получается обязательный лаг величиной от 5 до 50 мс, в зависимости от скорости порта, и это родовой порок. К тому же, внутри библиотек общего назначения поддержка портов официально есть до 115200 бод, остальное вне спецификаций и требует специально сертифицированных адаптеров (таких, как например, у Slamtec).

commit f1ec2813 (2022-07-27): Qt namespace workaround
	- Library with fully implemented Qt namespace workaround
	- Closes #9
	- Prevalidation of WT901 controller application