ElettraSciComp/witmotion_IMU_ros #10: Rviz. No messages received after 0.7.15-alpha.

tohax, 2022-08-29:
wt901c. Все версии до 0.7.15-alpha включительно работают. На следующих версиях Rviz говорит, что сообщения не получены.

twdragon, 2022-08-29:
Это говорит RViz или нода? Первым делом проверьте конфигурацию сборки сообщения IMU. Возможно, в новой версии примера конфигурационного файла не совпадают какие-то параметры, и сообщение IMU не собирается. Особенно это касается `use_native_orientation`, для этого надо через приложение-контроллер включить передачу ориентации в форме кватерниона

twdragon, 2022-08-29:
Я только что тестировал WT901 в бескорпусном исполнении, с параметром `use_native_orientation` IMU сообщение публикуется только с включенным аппаратно кватернионом.

twdragon, 2022-08-29:
@tohax а что в логе `rostopic echo`?

tohax, 2022-08-30:
use_native_orientation: false решило проблему. Настраивать кватернион еще не пробовал. Благодарю. Issue можно закрывать.

closed 2022-08-30 by twdragon