ElettraSciComp/witmotion_IMU_ros #11: Configuration quesiton: Scale IMU Z ?

rlabs-oss, 2022-08-30:
Is there any configuration parameter available to scale the IMU z linear_acceleration?
 
Lidar IMU: 
angular_velocity: 
  x: 0.7692424058914185
  y: -0.06967184692621231
  z: -0.1747824251651764

Wit_Motion IMU:
linear_acceleration: 
  x: -0.8622070550918579
  y: 0.2490820288658142
  z: 9.920170783996582

In the example above it should match the Lidar, and thus multiplied by 0.1x

twdragon, 2022-08-30:
Why do I see here the `angular_velocity` section? If acceleration is integrated by time, there is no need to adjust any bias, just select a proper measurement. Otherwise, it seems that your LIDAR accelerometer does not multiply the gravity axis by 0.1 but initially subtracts $9.81 m/s^2$ from Z axis measurement. Try to remove the gravity compensation setting from the LIDAR's IMU. In the worst case, the ROS driver itself does not provide the possibility to incorporate bias, so you need to use the controller application provided with the underlying library to adjust the hardware-dependent bias. It can be done using the key `--set-acceleration-bias X:Y:Z` where X, Y and Z are biases per axis represented as signed rationals.

However, I cannot recommend adjusting the hardware bias unless you are totally sure that your LIDAR accelerometer does not subtract gravity acceleration. Instead of this, it seemed quite better to write the simplest Python node script republishing IMU messages with proper adjustment, and include it in your launchfile.

rlabs-oss, 2022-08-30:
angular_velocity was posted by mistake, I meant to compare the acceleration - from the data below /livox/imu gravity is being subtracted?

section from rostopic echo /livox/imu
header: 
  seq: 2568
  stamp: 
    secs: 47
    nsecs: 296579600
  frame_id: "livox_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.04060148820281029
  y: -0.014278094284236431
  z: -0.1470855474472046
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.08941924571990967
  y: 0.000529848039150238
  z: 1.1143079996109009
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---

and rostopic echo /imu  (the witmotion IMU)
---
header: 
  seq: 4446
  stamp: 
    secs: 1661870547
    nsecs: 615156236
  frame_id: "imu"
orientation: 
  x: 0.004658230741860487
  y: 0.035490425767317695
  z: -0.08222356258143394
  w: 0.9959708913023649
orientation_covariance: [0.01899999938905239, 0.0, 0.0, 0.0, 0.012000000104308128, 0.0, 0.0, 0.0, 0.010700000450015068]
angular_velocity: 
  x: 0.001065264455974102
  y: 0.0
  z: 0.0
angular_velocity_covariance: [0.11990000307559967, 0.0, 0.0, 0.0, 0.5752999782562256, 0.0, 0.0, 0.0, 0.02669999934732914]
linear_acceleration: 
  x: -0.8286767601966858
  y: 0.3544628918170929
  z: 9.5896577835083
linear_acceleration_covariance: [0.008799999952316284, 0.0, 0.0, 0.0, 0.054999999701976776, 0.0, 0.0, 0.0, 0.02669999934732914]
---

After attempting IMU to Lidar calibration (after subtracting gravity from linear_acceleration Z)
[Refinement]  Online Refinement 100% [||||||||||||||||||||||||||||||]

[Final Result] Rotation LiDAR to IMU    = 179.269126 -69.520855 164.556532 deg
[Final Result] Translation LiDAR to IMU = -0.450945  0.258602 -0.187264 m
[Final Result] Time Lag IMU to LiDAR    = 1661874299.99544144 s ????
[Final Result] Bias of Gyroscope        =  1.378125 -0.612872 -1.593472 rad/s
[Final Result] Bias of Accelerometer    =  0.034958 -0.009180 -0.023757 m/s^2
[Final Result] Gravity in World Frame   = -4.431547 -3.114117 -8.146427 m/s^2

twdragon, 2022-08-30:
1. You have the wrong timestamp on your LIDAR message: it is indicating the uptime of your LIDAR node instead of being set to `ros::Time::now()` value copying the current system timestamp.
2. The rotational origin of the Z axis of your LIDAR accelerometer is unknown, so the fusion here in absolute angles will never match.
3. Accelerometer measurements are seeming matching pretty well.
4. LIDAR accelerometer on the Z axis seems not calibrated in the gravity field (the measurement should be ~0 or ~9.81). Can you refer to the official LIDAR documentation to calibrate it? In case I think, it subtracts the gravity as well.

Thus, I can recommend you at first to review thoroughly the value set for fusion and remove absolute Euler angles from it. Which utility do you use for measurement fusion?

rlabs-oss, 2022-08-30:
Thank you for the analysis - it helps a lot. 
The package is: https://github.com/hku-mars/LiDAR_IMU_Init
The Livox Avia is using the BMI055 - if the Avia frames are sent to FASTER_LIO the package works as expected. 
Since the Livox implementation seems to be Livox specific I'm attempting to adapt the WitMotion IMU frames to match in order to use FASTER_LIO with the Mid40 Lidar which does not have an internal IMU.

closed 2022-09-06 by twdragon