ElettraSciComp/witmotion_IMU_ros #16: Test reports for ROS1 version on different sensors

twdragon, 2022-11-14:
This issue is dedicated to placing the test reports of the ROS1 version of this driver. All contributors are encouraged to place the test reports they have for this driver on the different Witmotion sensors here. Please link here also the other issues containing test results and experience feedback under the following template:

> ### Tested sensor
> <sensor model, case type, connection type>
> #### List of tested modes
> - list or table of tested baudrates,
> - frequencies,
> - polling rates,
> #### Maximal duration of stable working (approx.)
> #### Encountered issues and crashes
> <with logs if it is possible>

## ROS2 issue: #15

elpimous, 2023-01-02:
### **HWT905 TTL, with original usb adapter**

joewong00, 2023-05-18:
## WTGAHRS1 GPS IMU

Manual: https://m.media-amazon.com/images/I/81dW2zuBOeL.pdf
[image]

### Machine: Nvidia Jetson Xavier NX, Jetpack 4.6, ARM64 architecture, Ubuntu 18.04 & ROS Melodic

## Test Report

| Status      |  Baud Rate | Polling Interval | Description |
| ----------- | ----------- | ---------- | ---------- |
| Fail | 9600 | 10 | Launch success, no message output when rostopic echo |
| Fail | 9600 | 50 | Launch success, no message output when rostopic echo |  
| Fail | 57600 | 10 | Launch failed, process has died |  
| Fail | 57600 | 50 | Launch failed, process has died |  
| Fail | 115200 | 10 | Launch success, no message output when rostopic echo |  
| Fail | 115200 | 50 | Launch success, no message output when rostopic echo |

twdragon, 2023-05-18:
@joewong00 did you check the logs and `use_native_orientation` setting?

joewong00, 2023-08-04:
@twdragon sorry for the late reply. I tried it recently with the latest code and it's working on the model WTGAHRS1 imu, i can get the topic data at any baudrate and polling interval depending on the settings of the IMU. However, I realised the values for the GPS latitude longitude are wrong. 

The correct value should be:
[image]

While the value I'm getting is:
[image]
 
Thank you.

twdragon, 2023-08-04:
@joewong00 I need your help! Could you please assist me with what kind of data tuning we need? I cannot test on the actual GPS device, so could you possibly figure out the coefficient with which the wrong data can be converted to the correct (I guess it should be a constant)?

joewong00, 2023-08-07:
@twdragon I fixed it, the datasheet requires a floor division for the degree. So basically I have to do a slight modification on these 2 lines of code, and I'm getting the correct latitude and longitude data now.

[image]

twdragon, 2023-08-07:
@joewong00 OK, thanks a lot! I will prepare an self-assessed PR to fix this calibration issue. 

May I also mention your nickname on ROS wiki in the "Community Tested" section?

joewong00, 2023-08-07:
@twdragon sure, no problem.

twdragon, 2023-08-07:
https://github.com/ElettraSciComp/witmotion_IMU_QT/pull/9 merged, I will update this repository very soon with new submodule states. Afterwards, @joewong00 could you please prepare the new test report with new code and place it here?

joewong00, 2023-08-07:
> [ElettraSciComp/witmotion_IMU_QT#9](https://github.com/ElettraSciComp/witmotion_IMU_QT/pull/9) merged, I will update this repository very soon with new submodule states. Afterwards, @joewong00 could you please prepare the new test report with new code and place it here?

Sure no problem, update me when the changes are done.

joewong00, 2023-08-09:
@twdragon I pulled the new code and tested it, the GPS latitude and longitude are still wrong, we need to add the floor function to it as shown (in `witmotion_ros.cpp`):

[image]

then it works :). I'll come out with a test report afterwards. By the way, my name is wrongly spelled in ROS WIKI, it should be Qing Joe Wong not Quing Joe Wong. Thanks.

twdragon, 2023-08-09:
@joewong00 thank you! Typo fixed, I also fixed the function in the code, then I am preparing now a release!

joewong00, 2023-08-14:
Sure no problem. I will produce the test report soon.

joewong00, 2023-08-14:
@twdragon I've updated my previous comment on the test report, I'm going to do the test for ros2 version. You can also 
delete all of these comments, thanks.