ElettraSciComp/witmotion_IMU_ros #19: Trouble interfacing with Raspberry Pi

adarshrs, 2022-11-19:
I am running ROS Noetic on a Raspberry Pi and am trying to interface WT901 with this node. I have the IMU connected to the serial ports on the Raspberry Pi (tx, rx -> rx, tx) but the node cannot communicate with the IMU on the correct serial port and baud rate. I communicated with the IMU over I2C on the Raspberry Pi using a custom driver and UART on Arduino using sample code on the witmotion website. Is there something I am missing with respect to getting this ROS driver to work on a Raspberry Pi?

twdragon, 2022-11-20:
@adarshrs at first, check if your user is added to `dialout` group. Then try to see the symbols coming from the sensor in `minicom` or `screen` emulator. If you can see the symbols, then it is a library fault, and we will be in need of logs here to investigate.

adarshrs, 2022-11-22:
I have verified that the user is added. I am also able to communicate with the raspberry pi over the serial port from an arduino, so it looks like serial communication is working. However, I see no output from the sensor in Picocom on the raspberry pi. I am still able to see output from the sensor using an arduino

twdragon, 2022-11-22:
As you described that you can see the Raspberry console from Arduino, here is the problem: the UART transceiver of Raspberry could be monopolized for being used as a serial console port. You should run the Raspberry Pi configuration utility and change the port setting not to redirect the system console TTY to it.

adarshrs, 2022-11-22:
I have done this configuration already, enabling the serial hardware but disabling the console. To clarify, I could not see the raspberry pi console on the arduino. I was able to send string messages from the arduino to the raspberry pi

twdragon, 2022-11-22:
@adarshrs did you test duplex data transfer? On which baudrate and mode does your port work? Did you try to play with the controller applications of the underlying library to determine the exact baudrate and mode? Also, did you try to listen on `picocom` using different baudrates? Maybe, there is still a mismatch?

adarshrs, 2022-11-22:
Update: I was able to see symbols on picocom with the IMU connected. Not sure what changed, but it looks like the IMU is sending data. The ROS driver doesn't still work, though. Here are the logs when I try to launch:

# Rosout:
```
$ roslaunch witmotion_ros wt901.launch
... logging to /home/pi/.ros/log/7c0a9c82-6a9c-11ed-96a2-b827eb3cf8c3/roslaunch-raspberrypi-2059.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:38309/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /witmotion_imu/altimeter_publisher/addition: 0.0
 * /witmotion_imu/altimeter_publisher/coefficient: 1.0
 * /witmotion_imu/altimeter_publisher/enabled: False
 * /witmotion_imu/altimeter_publisher/topic_name: /altitude
 * /witmotion_imu/barometer_publisher/addition: 0.0
 * /witmotion_imu/barometer_publisher/coefficient: 1.0
 * /witmotion_imu/barometer_publisher/enabled: False
 * /witmotion_imu/barometer_publisher/frame_id: base_link
 * /witmotion_imu/barometer_publisher/topic_name: /barometer
 * /witmotion_imu/barometer_publisher/variance: 0.001
 * /witmotion_imu/baud_rate: 9600
 * /witmotion_imu/gps_publisher/enabled: False
 * /witmotion_imu/gps_publisher/ground_speed_topic_name: /gps_ground_speed
 * /witmotion_imu/gps_publisher/navsat_altitude_topic_name: /gps_altitude
 * /witmotion_imu/gps_publisher/navsat_fix_frame_id: world
 * /witmotion_imu/gps_publisher/navsat_fix_topic_name: /gps
 * /witmotion_imu/gps_publisher/navsat_satellites_topic_name: /gps_satellites
 * /witmotion_imu/gps_publisher/navsat_variance_topic_name: /gps_variance
 * /witmotion_imu/imu_publisher/frame_id: imu
 * /witmotion_imu/imu_publisher/measurements/acceleration/covariance: [0.0364, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/acceleration/enabled: True
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/covariance: [0.0663, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/enabled: True
 * /witmotion_imu/imu_publisher/measurements/orientation/covariance: [0.0479, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/orientation/enabled: True
 * /witmotion_imu/imu_publisher/topic_name: /imu
 * /witmotion_imu/imu_publisher/use_native_orientation: True
 * /witmotion_imu/magnetometer_publisher/addition: 0.0
 * /witmotion_imu/magnetometer_publisher/coefficient: 1e-08
 * /witmotion_imu/magnetometer_publisher/covariance: [1.87123e-07, 0, ...
 * /witmotion_imu/magnetometer_publisher/enabled: True
 * /witmotion_imu/magnetometer_publisher/frame_id: compass
 * /witmotion_imu/magnetometer_publisher/topic_name: /magnetometer
 * /witmotion_imu/orientation_publisher/enabled: True
 * /witmotion_imu/orientation_publisher/topic_name: /orientation
 * /witmotion_imu/polling_interval: 10
 * /witmotion_imu/port: /dev/ttyS0
 * /witmotion_imu/restart_service_name: /restart_imu
 * /witmotion_imu/rtc_publisher/enabled: False
 * /witmotion_imu/rtc_publisher/topic_name: /witmotion_clock
 * /witmotion_imu/temperature_publisher/addition: 0.0
 * /witmotion_imu/temperature_publisher/coefficient: 1.0
 * /witmotion_imu/temperature_publisher/enabled: True
 * /witmotion_imu/temperature_publisher/frame_id: base_link
 * /witmotion_imu/temperature_publisher/from_message: magnetometer
 * /witmotion_imu/temperature_publisher/topic_name: /temperature
 * /witmotion_imu/temperature_publisher/variance: 0.01829

NODES
  /
    witmotion_imu (witmotion_ros/witmotion_ros_node)

[... 117 lines cut ...]
[roslaunch.pmon][INFO] 2022-11-22 14:33:14,860: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 1950889024)>: remaining procs are [<roslaunch.nodeprocess.LocalProcess object at 0x7449e510>, <roslaunch.nodeprocess.LocalProcess object at 0x7449ea30>]
[roslaunch.pmon][INFO] 2022-11-22 14:33:14,866: ProcessMonitor exit: killing rosout-1
[roslaunch][INFO] 2022-11-22 14:33:14,867: [rosout-1] killing on exit
[roslaunch][INFO] 2022-11-22 14:33:14,868: process[rosout-1]: killing os process with pid[2079] pgid[2079]
[roslaunch][INFO] 2022-11-22 14:33:14,869: [rosout-1] sending SIGINT to pgid [2079]
[roslaunch][INFO] 2022-11-22 14:33:14,869: [rosout-1] sent SIGINT to pgid [2079]
[roslaunch][INFO] 2022-11-22 14:33:15,271: process[rosout-1]: SIGINT killed with return value 0
[roslaunch.pmon][INFO] 2022-11-22 14:33:15,272: ProcessMonitor exit: killing master
[roslaunch][INFO] 2022-11-22 14:33:15,272: [master] killing on exit
[roslaunch][INFO] 2022-11-22 14:33:15,272: process[master]: killing os process with pid[2068] pgid[2068]
[roslaunch][INFO] 2022-11-22 14:33:15,273: [master] sending SIGINT to pgid [2068]
[roslaunch][INFO] 2022-11-22 14:33:15,273: [master] sent SIGINT to pgid [2068]
[roslaunch][INFO] 2022-11-22 14:33:15,574: process[master]: SIGINT killed with return value 0
[roslaunch.pmon][INFO] 2022-11-22 14:33:15,579: ProcessMonitor exit: cleaning up data structures and signals
[roslaunch.pmon][INFO] 2022-11-22 14:33:15,579: ProcessMonitor exit: pmon has shutdown
[roslaunch][INFO] 2022-11-22 14:33:15,579: process monitor is done spinning, initiating full shutdown
[roslaunch][INFO] 2022-11-22 14:33:15,580: runner.stop()
[roslaunch][INFO] 2022-11-22 14:33:15,580: shutting down processing monitor...
[roslaunch][INFO] 2022-11-22 14:33:15,581: shutting down processing monitor <ProcessMonitor(ProcessMonitor-1, stopped daemon 1950889024)>
[roslaunch.pmon][INFO] 2022-11-22 14:33:15,581: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 1950889024)>
[roslaunch][INFO] 2022-11-22 14:33:15,581: ... shutting down processing monitor complete
[roslaunch][INFO] 2022-11-22 14:33:15,582: done
[roslaunch.pmon][INFO] 2022-11-22 14:33:15,582: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 1950889024)>
[rospy.core][INFO] 2022-11-22 14:33:15,583: signal_shutdown [atexit]
```

twdragon, 2022-11-23:
@adarshrs the log says:
```
Instantiating timer at 10 ms
[ERROR] [1669145590.937285603]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1669145590.943398563]: Entering SUSPENDED state
QObject::killTimer: Timers cannot be stopped from another thread
QObject::~QObject: Timers cannot be stopped from another thread
QSocketNotifier: Socket notifiers cannot be enabled or disabled from another thread
QSocketNotifier: Invalid socket 14 and type 'Read', disabling...
[witmotion_imu-2] process has died [pid 2086, exit code -11, cmd /home/pi/aerobat_ws/devel/lib/witmotion_ros/witmotion_ros_node __name:=witmotion_imu __log:=/home/pi/.ros/log/7c0a9c82-6a9c-11ed-96a2-b827eb3cf8c3/witmotion_imu-2.log].
log file: /home/pi/.ros/log/7c0a9c82-6a9c-11ed-96a2-b827eb3cf8c3/witmotion_imu-2*.log
```

It means that the data are not really already thrown to the port while the node tries to call `read()` to acquire it. Please check also the measurement rate of your sensor and try to change `polling_interval` parameter to 50 or 100 ms to see if the data are really written to the port by the sensor. If the sensor's measurement rate is too low, you should use the controller application from the underlying library to tune it. Also please try to determine which  exact types of the messages your sensor generates with `message_enumerator` application from the underlying library.

adarshrs, 2022-11-23:
@twdragon I tried out different baud rates and polling frequencies. According to the data sheet, the default baud rate is 9600. I ran message enumeration and this is the output:

```

Press Ctrl+C to stop enumeration and see the report
Opening device
'ttyS0" at 9600 baud
Instantiating timer at 20 ms
^C
Closing TTL connection
WITMOTION UART MESSAGE ENUMERATOR BY TWDRAGON
Acquired at Wednesday, 23 November 2022 14:57:30 EST

ID | Qty | Description
0x51 | 958 | Accelerations
0x52 | 958 | Angular velocities
0x53 | 958 | Spatial orientation (Euler angles)
0x54 | 958 | Magnetometer/Magnetic orientation

Unknown IDs: 0 [ ]
Total messages: 3832

```

This does not produce the error, but no messages are published either. This is the output of the roslaunch:

```
auto-starting new master
process[master]: started with pid [1917]
ROS_MASTER_URI=http://Localhost:11311
setting /run_id to 2b8487f0-6b69-11ed-bad5-b827eb3cf8c3
process[rosout-1]: started with pid [1928]
started core service [/rosout]
process[witmotion_imu-2]: started with pid [1935]
Opening device "ttySO" at 9600 baud
Instantiating timer at 20 mS
```

There is no output after this and no messages published.

twdragon, 2022-11-25:
@adarshrs I did not see before: your sensor does not provide Quaternion orientation whilst ` imu_publisher/use_native_orientation` parameter is set to `true`. In this case, the orientation section of the IMU message is never considered completed and the message is never published. Set this parameter to `false` and the node will start working. 

This situation is explained [here](https://wiki.ros.org/witmotion_ros#Troubleshooting) in the node documentation.

twdragon, 2022-11-25:
@adarshrs if you will present here the positive result of your test, I will close this issue as solved.

adarshrs, 2022-11-28:
@twdragon Thank you for your suggestion. Turning off native orientation worked and I am able to get /imu messages at 10Hz. I tried using the underlying library to increase the polling rate using the following command:

`rosrun witmotion_ros witmotionctl-wt901 -d /dev/ttyS0 --set-frequency 100`

The output suggests that this works, however, I get a segmentation fault when trying to read at a polling interval of 5 ms  (100 Hz). At lower frequency 50 Hz, I am seeing messages published at a rate of 16 Hz. 

Is this a hardware limitation or is there some way of getting higher publish rates? I am looking for rates of at least 100 Hz, ideally 200 Hz.

twdragon, 2022-11-29:
@adarshrs as it is described [here](https://wiki.ros.org/witmotion_ros#Actual_bandwidth_and_measurement_frequencies) there is a limitation due to the sensor may *throw the packets* on the given **frequency**, but the ROS IMU message is built from at least 2 Witmotion packets. There is also a synchronization limitation due to the ratio between the selected baudrate and measurement frequency. The highest measurement frequencies, obviously, are accessible only on the highest baudrates. I heard some use cases (e.a @tohax) with actual publishing frequencies up to 160 Hz, so the limitation exists.

In your case I can advice you to use the commands:
```
witmotionctl-wt901 -d /dev/ttyS0 --set-baudrate 115200
witmotionctl-wt901 -d /dev/ttyS0 --set-frequency 200
witmotionctl-wt901 -d /dev/ttyS0 -i 7
```
to set the packet throw interval to 5ms, as it is described [here](https://elettrascicomp.github.io/witmotion_IMU_QT/dc/ddc/namespacewitmotion.html#a2a0798fb689894f2b7f998303bd8a4b3a6b05a7147ce284d6ef2e1ea5fca0dc4b), and to check if the 7ms polling interval works, otherwise, try to increase it. Then try to play with the actual **port polling** interval using the `polling_interval` parameter to determine the polling frequency feasible for your hardware setup. There is also a kernel polling rate limitation that depends on the architecture and may vary between builds. You should select the value on which your setup would be stable.

As the device is now working, I will close this issue as completed. I will be very appreciated if you would write a test report for Raspberry setup in the [dedicated issue](https://github.com/ElettraSciComp/witmotion_IMU_ros/issues/16).

closed 2022-11-29 by twdragon