ElettraSciComp/witmotion_IMU_ros #20: I can see topics, but no values on hwt905-ttl ?!

elpimous, 2023-01-01:
Hello, I bought your hwt905, ttl version.
Calibration on win10 ok.
restarted under ubuntu 20, under ros noetic
When i roslaunch : `roslaunch witmotion_ros witmotion.launch`,
imu seems to work : 

`roslaunch witmotion_ros witmotion.launch
... logging to /home/ylo2/.ros/log/3bbc653e-8a1d-11ed-9ad3-253eccaef672/roslaunch-ylo2-UP-WHL01-10499.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/ylo2/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://ylo2-UP-WHL01:36477/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /witmotion_imu/altimeter_publisher/addition: 0.0
 * /witmotion_imu/altimeter_publisher/coefficient: 1.0
 * /witmotion_imu/altimeter_publisher/enabled: True
 * /witmotion_imu/altimeter_publisher/topic_name: /altitude
 * /witmotion_imu/barometer_publisher/addition: 0.0
 * /witmotion_imu/barometer_publisher/coefficient: 1.0
 * /witmotion_imu/barometer_publisher/enabled: True
 * /witmotion_imu/barometer_publisher/frame_id: base_link
 * /witmotion_imu/barometer_publisher/topic_name: /barometer
 * /witmotion_imu/barometer_publisher/variance: 0.001
 * /witmotion_imu/baud_rate: 115200
 * /witmotion_imu/gps_publisher/enabled: False
 * /witmotion_imu/gps_publisher/ground_speed_topic_name: /gps_ground_speed
 * /witmotion_imu/gps_publisher/navsat_altitude_topic_name: /gps_altitude
 * /witmotion_imu/gps_publisher/navsat_fix_frame_id: world
 * /witmotion_imu/gps_publisher/navsat_fix_topic_name: /gps
 * /witmotion_imu/gps_publisher/navsat_satellites_topic_name: /gps_satellites
 * /witmotion_imu/gps_publisher/navsat_variance_topic_name: /gps_variance
 * /witmotion_imu/imu_publisher/frame_id: imu
 * /witmotion_imu/imu_publisher/measurements/acceleration/covariance: [0.0088, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/acceleration/enabled: True
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/covariance: [0.1199, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/enabled: True
 * /witmotion_imu/imu_publisher/measurements/orientation/covariance: [0.019, 0, 0, 0, ...
 * /witmotion_imu/imu_publisher/measurements/orientation/enabled: True
 * /witmotion_imu/imu_publisher/topic_name: /imu
 * /witmotion_imu/imu_publisher/use_native_orientation: False
 * /witmotion_imu/magnetometer_publisher/addition: 0.0
 * /witmotion_imu/magnetometer_publisher/coefficient: 1e-08
 * /witmotion_imu/magnetometer_publisher/covariance: [1.87123e-07, 0, ...
 * /witmotion_imu/magnetometer_publisher/enabled: True
 * /witmotion_imu/magnetometer_publisher/frame_id: imu
 * /witmotion_imu/magnetometer_publisher/topic_name: /magnetometer
 * /witmotion_imu/orientation_publisher/enabled: True
 * /witmotion_imu/orientation_publisher/topic_name: /orientation
 * /witmotion_imu/polling_interval: 150
 * /witmotion_imu/port: ttyUSB0
 * /witmotion_imu/restart_service_name: /restart_imu
 * /witmotion_imu/rtc_publisher/enabled: True
 * /witmotion_imu/rtc_publisher/presync: True
 * /witmotion_imu/rtc_publisher/topic_name: /witmotion_clock
 * /witmotion_imu/temperature_publisher/addition: 0.0
 * /witmotion_imu/temperature_publisher/coefficient: 1.0
 * /witmotion_imu/temperature_publisher/enabled: True
 * /witmotion_imu/temperature_publisher/frame_id: base_link
 * /witmotion_imu/temperature_publisher/from_message: magnetometer
 * /witmotion_imu/temperature_publisher/topic_name: /temperature
 * /witmotion_imu/temperature_publisher/variance: 0.02683

NODES
  /
    witmotion_imu (witmotion_ros/witmotion_ros_node)

auto-starting new master
process[master]: started with pid [10514]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3bbc653e-8a1d-11ed-9ad3-253eccaef672
process[rosout-1]: started with pid [10531]
started core service [/rosout]
process[witmotion_imu-2]: started with pid [10538]
Opening device "ttyUSB0" at 115200 baud
[ INFO] [1672609368.553916924]: Initiating RTC pre-synchonization: current timestamp 2023-01-01T22:42:48.553
[ INFO] [1672609368.554407922]: Configuration ROM: lock removal started
Instantiating timer at 150 ms
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x69
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x33
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x32
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x31
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x30
Configuration packet sent, flushing buffers...
Configuration completed
[ INFO] [1672609373.555362664]: RTC pre-synchonization completed, saving configuration
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x0
Configuration packet sent, flushing buffers...
Configuration completed
[ INFO] [1672609374.555630283]: RTC synchronized`

Howerver, whe I run : `rostopic echo /altitude or rostopic echo /imu`, I don't see any values ?!!

Any idea ?

twdragon, 2023-01-02:
@elpimous try to run at first
```bash
rosrun witmotion_ros message_enumerator
```
then you will determine the exact set of messages your sensor produces. You should set the IMU measurement elements' `enabled` parameters **exact matching** the elements your sensor produces. Please refer to the [`message_enumerator` application documentation](https://wiki.ros.org/witmotion_ros#Message_Enumerator) to see the options.

If you will not see any messages, try to play with `baud_rate` parameter. It is the chance that the default baud rate and polling frequency of the sensor do not match the values you set in the node configuration.

twdragon, 2023-01-02:
@elpimous as your sensor was never tested in ROS, all the contributors will be appreciated if you would write the test/experience report in the dedicated issue #16

elpimous, 2023-01-02:
@twdragon .
Hi Andrei, tested `rosrun witmotion_ros message-enumerator -d ttyUSB0 -bxxx -pxxx`
ttyUSB0 is the correct port,
tested with baud rate : 4800, 9600, 115200, 961200
tested each, with Instantiating timer : 5, 10, 20, ... 200. (each +10)
`
WITMOTION UART MESSAGE ENUMERATOR BY TWDRAGON

Acquired at lundi 2 janvier 2023 14:48:32 CET

ID	Qty	Description

	Unknown IDs: 0 [  ] 
Total messages: 0
`
Seems impossible to make recognized to ros1

elpimous, 2023-01-02:
I'm a bit disappointed, because I was looking for a Ros ready IMU (like my previous Myahrs+ one)

twdragon, 2023-01-02:
@elpimous is your user in `dialout` group? Did you try to see the bytes the sensor emits on `minicom` or `screen`?

elpimous, 2023-01-02:
1/  is your user in dialout group : **yes**
```
ylo2@ylo2-UP-WHL01:~$ sudo adduser ylo2 dialout
L'utilisateur « ylo2 » appartient déjà au groupe « dialout »
```

twdragon, 2023-01-02:
@elpimous It seems that the port settings in Linux for `message-enumerator` are still incorrect. Try to do the following:
- Thoroughly review the communication settings you set for the sensor in the Windows program: baud rate, measurement frequency, data format;
- The default baud rate for `minicom` is 115200 baud for the desktop system. Try to determine which baudrate (using the `minicom -b <rate>` command) allows you to see the bytes. It is also could be done using the `minicom` configuration menu;
- Review the types of data your sensor produces (sometimes magnetometer and inclinometer are switched off by default)
- Review the `minicom` settings to get the exact port parameters;
- Try again `message-enumerator` with exact values of the port parameters set until you will see the list of messages.

You see the bytes on the `minicom` output, so the sensor throws some data, but the controller application cannot read them. It can happen due to the following:
- Malformed protocol on the sensor's side (GPS instead of IMU);
- Wrong baud rate;
- Too small polling interval (start with 50ms, then try to increase or decrease, 150ms you have does not seem convenient);
- Wrong measurement mode (for example, one-shot mode set using official Windows controller);
- Hardware failure (for example, TTL-USB converter does not support the selected baudrate).

twdragon, 2023-01-02:
@elpimous the baudrate indicated is `921600` baud, it is the non-standard value that is not supported by `libqt5serial`. Try to change to 115200 baud, then repeat an experiment with `message-enumerator`.

elpimous, 2023-01-02:
Hello, SUCCESS !
changed in windows10 : 

- Baud changed to 115200, 
- bandwidth & output_rate to 10Hz

elpimous, 2023-01-02:
ros command : 

`ylo2@ylo2-UP-WHL01:~$ roslaunch witmotion_ros witmotion.launch
... logging to /home/ylo2/.ros/log/a002618c-8ae1-11ed-b48c-adb2194ffe95/roslaunch-ylo2-UP-WHL01-7702.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ylo2-UP-WHL01:40597/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /witmotion_imu/altimeter_publisher/addition: 0.0
 * /witmotion_imu/altimeter_publisher/coefficient: 1.0
 * /witmotion_imu/altimeter_publisher/enabled: False
 * /witmotion_imu/altimeter_publisher/topic_name: /altitude
 * /witmotion_imu/barometer_publisher/addition: 0.0
 * /witmotion_imu/barometer_publisher/coefficient: 1.0
 * /witmotion_imu/barometer_publisher/enabled: False
 * /witmotion_imu/barometer_publisher/frame_id: base_link
 * /witmotion_imu/barometer_publisher/topic_name: /barometer
 * /witmotion_imu/barometer_publisher/variance: 0.001
 * /witmotion_imu/baud_rate: 115200
 * /witmotion_imu/gps_publisher/enabled: False
 * /witmotion_imu/gps_publisher/ground_speed_topic_name: /gps_ground_speed
 * /witmotion_imu/gps_publisher/navsat_altitude_topic_name: /gps_altitude
 * /witmotion_imu/gps_publisher/navsat_fix_frame_id: world
 * /witmotion_imu/gps_publisher/navsat_fix_topic_name: /gps
 * /witmotion_imu/gps_publisher/navsat_satellites_topic_name: /gps_satellites
 * /witmotion_imu/gps_publisher/navsat_variance_topic_name: /gps_variance
 * /witmotion_imu/imu_publisher/frame_id: imu
 * /witmotion_imu/imu_publisher/measurements/acceleration/covariance: [0.0088, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/acceleration/enabled: True
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/covariance: [0.1199, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/enabled: True
 * /witmotion_imu/imu_publisher/measurements/orientation/covariance: [0.019, 0, 0, 0, ...
 * /witmotion_imu/imu_publisher/measurements/orientation/enabled: True
 * /witmotion_imu/imu_publisher/topic_name: /imu
 * /witmotion_imu/imu_publisher/use_native_orientation: False
 * /witmotion_imu/magnetometer_publisher/addition: 0.0
 * /witmotion_imu/magnetometer_publisher/coefficient: 1e-08
 * /witmotion_imu/magnetometer_publisher/covariance: [1.87123e-07, 0, ...
 * /witmotion_imu/magnetometer_publisher/enabled: True
 * /witmotion_imu/magnetometer_publisher/frame_id: imu
 * /witmotion_imu/magnetometer_publisher/topic_name: /magnetometer
 * /witmotion_imu/orientation_publisher/enabled: True
 * /witmotion_imu/orientation_publisher/topic_name: /orientation
 * /witmotion_imu/polling_interval: 150
 * /witmotion_imu/port: ttyUSB0
 * /witmotion_imu/restart_service_name: /restart_imu
 * /witmotion_imu/rtc_publisher/enabled: True
 * /witmotion_imu/rtc_publisher/presync: True
 * /witmotion_imu/rtc_publisher/topic_name: /witmotion_clock
 * /witmotion_imu/temperature_publisher/addition: 0.0
 * /witmotion_imu/temperature_publisher/coefficient: 1.0
 * /witmotion_imu/temperature_publisher/enabled: True
 * /witmotion_imu/temperature_publisher/frame_id: base_link
 * /witmotion_imu/temperature_publisher/from_message: magnetometer
 * /witmotion_imu/temperature_publisher/topic_name: /temperature
 * /witmotion_imu/temperature_publisher/variance: 0.02683

NODES
  /
    witmotion_imu (witmotion_ros/witmotion_ros_node)

auto-starting new master
process[master]: started with pid [7717]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a002618c-8ae1-11ed-b48c-adb2194ffe95
process[rosout-1]: started with pid [7734]
started core service [/rosout]
process[witmotion_imu-2]: started with pid [7741]
Opening device "ttyUSB0" at 115200 baud
[ INFO] [1672693718.135335866]: Initiating RTC pre-synchonization: current timestamp 2023-01-02T22:08:38.135
[ INFO] [1672693718.135874375]: Configuration ROM: lock removal started
Instantiating timer at 150 ms
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x69
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x33
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x32
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x31
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x30
Configuration packet sent, flushing buffers...
Configuration completed
[ INFO] [1672693723.136747909]: RTC pre-synchonization completed, saving configuration
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x0
Configuration packet sent, flushing buffers...
Configuration completed
[ INFO] [1672693724.137026460]: RTC synchronized`

rostopic : 

`ylo2@ylo2-UP-WHL01:~$ rostopic echo /imu
header: 
  seq: 4592
  stamp: 
    secs: 1672694175
    nsecs:  36364410
  frame_id: "imu"
orientation: 
  x: 0.014039034877760788
  y: -0.00643245922189693
  z: 0.9214943385342607
  w: 0.38808467248445305
orientation_covariance: [0.01899999938905239, 0.0, 0.0, 0.0, 0.012000000104308128, 0.0, 0.0, 0.0, 0.010700000450015068]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [0.11990000307559967, 0.0, 0.0, 0.0, 0.5752999782562256, 0.0, 0.0, 0.0, 0.02669999934732914]
linear_acceleration: 
  x: 0.30177247524261475
  y: -0.009580078534781933
  z: 9.786049842834473
linear_acceleration_covariance: [0.008799999952316284, 0.0, 0.0, 0.0, 0.054999999701976776, 0.0, 0.0, 0.0, 0.02669999934732914]
---
`

twdragon, 2023-01-02:
@elpimous COMPLIMENTS!

If you would prepare the test report for #16 I will be very appreciated!

I will close the issue a bit later as the completed task

elpimous, 2023-01-02:
@twdragon what is the maximum Hz could I try ?
Need the fastest, to control my quadruped robot (torque mode)
Of course, i'll report my test on #16

twdragon, 2023-01-02:
@elpimous thank you! 

I tested the existing modules on 115200 baud, 200 Hz measurement rate (that corresponds to appr. 180 Hz message drop rate). I suggest you start with a 100 Hz and 12ms polling rate. Then try to increase it to 125 and 200 Hz on a 10 ms polling rate, and decrease the polling rate to 5 only if all the previous attempts generate stable output without crashes.

closed 2023-01-03 by twdragon

twdragon, 2023-01-03:
Documented as a troubleshooting advice: https://wiki.ros.org/witmotion_ros#Non-standard_baudrates

SunilSrivatsav9, 2023-06-14:
How can we control the measurement rate in ROS package. Is there any parameter for measurement rate (frequency). I found only baud rate and Polling rate but didn't find the measurement rate.

twdragon, 2023-06-14:
@SunilSrivatsav9 The measurement rate should be set on the sensor directly because the IMU message contains 3 independent packets dropped by the sensor. So you need to use the controller application (JY901 one in the underlying library may be sufficient) to set the measurement rate. It is also internally dependent on the selected baudrate, so it cannot be adequately documented.