ElettraSciComp/witmotion_IMU_ros #21: wt901 sensor covariance

zejiekong, 2023-01-03:
Hi, how is the accelerometer, gyroscope ... covariance matrix obtained for the wt901 sensor? As shown in the wt901.yml config file. Thank you.

zejiekong, 2023-01-04:
What methods would you recommend to remove the noise and bias from the witmotion imu if it is possible?

twdragon, 2023-01-04:
@zejiekong the sensors have their own implementation of the Kalman filter (at least, Witmotion Ltd. declares it), so the attempts to eliminate the noise with some kind of smart filtering likely will provoke filter dissipation. I can recommend exchanging the measurement frequency for precision and making the accumulation register with loopback-based deviation calculation via the known existing noise covariance.

twdragon, 2023-01-04:
@zejiekong just using several subsequent measurements from the sensor to reformulate new IMU ROS message with refined values

zejiekong, 2023-01-06:
Also, I see there is a CalculateAccelerationBias method in the wt901-uart.cpp file. Does that calculate the accelerometer bias stability? And what does the SetAccelerationBias method do?

twdragon, 2023-01-06:
@zejiekong they are convenience methods to set up the constant biases on the linear accelerometer via controller application. The functionality is partially described [here](https://elettrascicomp.github.io/witmotion_IMU_QT/d2/da6/acceleration-bias.html), but the documentation is still WIP. These methods do not calculate stability biases or noise covariance.

closed 2023-01-10 by twdragon