ElettraSciComp/witmotion_IMU_ros #28: its not working for WT901BLECL5.0

Nareshkumar-g, 2023-06-12:
The sensor is connecting but its not outputting the data,
This is the error msg, i got
setting /run_id to ecf24ac8-090a-11ee-a325-0242c6588c83
process[rosout-1]: started with pid [8024]
started core service [/rosout]
process[witmotion_imu-2]: started with pid [8052]
Opening device "ttyUSB2" at 115200 baud
[ INFO] [1686565304.837663888]: Initiating RTC pre-synchonization: current timestamp 2023-06-12T15:51:44.837
Instantiating timer at 50 ms
[ INFO] [1686565304.842992995]: Configuration ROM: lock removal started
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x69
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x33
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x32
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x31
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x30
Configuration packet sent, flushing buffers...
Configuration completed
[ INFO] [1686565309.844640790]: RTC pre-synchonization completed, saving configuration
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x0
Configuration packet sent, flushing buffers...
Configuration completed
[ INFO] [1686565310.845034193]: RTC synchronized
[ERROR] [1686565310.845576275]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1686565310.845784884]: Entering SUSPENDED state
QObject::killTimer: Timers cannot be stopped from another thread
QObject::~QObject: Timers cannot be stopped from another thread
QSocketNotifier: Socket notifiers cannot be enabled or disabled from another thread
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1686565310.847763963]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1686565310.847867035]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1686565310.848724190]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1686565310.848851647]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1686565310.849020799]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1686565310.849107072]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!

twdragon, 2023-06-12:
What is an actual baudrate you have set up on the sensor? Could you access it and enumerate the messages with `message-enumerator` utility?

Nareshkumar-g, 2023-06-12:
The baudrate i set is 115200 as per the data sheet.

twdragon, 2023-06-12:
Yes, but was it set up also on the sensor? Can you see the data with this baudrate from another application? Did you try another values?

Nareshkumar-g, 2023-06-12:
Yeah sensor is working fine, i tries with the witmotion app provided by the manufacturer, its working well,
Does this package support WT901BLECL 5.0 IMU Sensor ?

twdragon, 2023-06-12:
@Nareshkumar-g It depends on whether the sensor:
- drops the data formatted by the [Witmotion 11-byte communication protocol](https://elettrascicomp.github.io/witmotion_IMU_QT/d6/de1/witmotion_protocol.html)
- The connection settings are set up properly

Please check the exact connection settings on your Witmotion controller application. It is a chance that you have wrongly set baudrate there, so the sensor remembers it and it is actually works not on 115200 baud.

SunilSrivatsav9, 2023-06-12:
I have the same issue. 
Initially I got the below error

*******     Error 1 ******************
Opening device "ttyUSB0" at 9600 baud
[ INFO] [1686612884.631942076]: Initiating RTC pre-synchonization: current timestamp 2023-06-13T08:34:44.631
[ INFO] [1686612884.632297483]: Configuration ROM: lock removal started
[ INFO] [1686612889.634066069]: RTC pre-synchonization completed, saving configuration
[ INFO] [1686612890.634352841]: RTC synchronized
[ERROR] [1686612890.634802042]: Sensor error: Error opening the port!
[ INFO] [1686612890.634899953]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!

I have tried to change the USB access permissions using the following command 
**sudo chmod a+rw /dev/ttyUSB0**

Then I got the following error:

process[witmotion_imu-1]: started with pid [11572]
Opening device "ttyUSB0" at 9600 baud
[ INFO] [1686614065.787490338]: Initiating RTC pre-synchonization: current timestamp 2023-06-13T08:54:25.787
[ INFO] [1686614065.787867738]: Configuration ROM: lock removal started
Instantiating timer at 50 ms
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x69
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x33
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x32
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x31
Configuration packet sent, flushing buffers...
Configuration completed
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x30
Configuration packet sent, flushing buffers...
Configuration completed
[ INFO] [1686614070.789366009]: RTC pre-synchonization completed, saving configuration
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x0
Configuration packet sent, flushing buffers...
Configuration completed
[ INFO] [1686614071.789645050]: RTC synchronized
[ERROR] [1686614071.790033951]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1686614071.790075629]: Entering SUSPENDED state
QObject::killTimer: Timers cannot be stopped from another thread
QObject::~QObject: Timers cannot be stopped from another thread
QSocketNotifier: Socket notifiers cannot be enabled or disabled from another thread
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1686614071.790718832]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1686614071.790753356]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1686614071.790782941]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1686614071.790816132]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1686614071.790884519]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1686614071.790927138]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1686614071.790955150]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1686614071.790981289]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1686614071.791038104]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1686614071.791067439]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1686614071.791111340]: Sensor error: No data acquired during last 3 iterations, please check the baudrate!
[ INFO] [1686614071.791137008]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!

twdragon, 2023-06-13:
The timeout parameter will be added soon. Now the question is the baudrate, we should ensure the value the sensor remembers by itself

twdragon, 2023-06-13:
@Nareshkumar-g then:
- Ensure the username of the system user is in `dialout` group
- Try to run `message-enumerator` to determine on which virtual filesystem name the device drops the data

If you share the output of `message-enumerator`'s output we will see what is not working properly

twdragon, 2023-06-13:
@SunilSrivatsav9 the output is for 9600 baud. Is it the same for 115200 and the other rates?

SunilSrivatsav9, 2023-06-13:
Yes its the same for all values.
I feel the issue is with interface (HW905- **RS485**)

SunilSrivatsav9, 2023-06-14:
@twdragon I am not using any driver. I directly connected the USB-SERIAL CH340 convertor to the system. I just checked the port using ls /dev/ttyUSB* and gave the root permissions to the port.  Its working using windows SDK, but its unable to receive data with Ubuntu ( ROS ).

SunilSrivatsav9, 2023-06-14:
Below are the Parameters from Windows SDK 
[image]
[image]

twdragon, 2023-06-14:
@SunilSrivatsav9 it seems your device got 9600 baud. Can you see anything connecting to the corresponding port via Linux on `minicom` with this baudrate?

SunilSrivatsav9, 2023-06-15:
I was successful in using the given link https://github.com/WITMOTION/WitStandardModbus_WT901C485
Since I am using RS485 communication, using Modbus protocol as mentioned in the above package helped in solving the issue.

closed 2023-12-29 by twdragon