ElettraSciComp/witmotion_IMU_ros #30: [ROS2] clean build failing on ROS2 Foxy

Chambana, 2023-06-19:
When I follow the installation instruction within a ROS2 **Foxy** environment on Ubuntu **20.04**, the colcon build command fails with hundreds of these errors:

```
CMake Error at /opt/ros/foxy/share/ament_cmake_export_libraries/cmake/ament_export_libraries.cmake:83 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_LOCATION" is not allowed.
Call Stack (most recent call first):
  CMakeLists.txt:98 (ament_export_libraries)
```
[build fail log.txt](https://github.com/ElettraSciComp/witmotion_IMU_ros/files/11793888/build.fail.log.txt)

I've attached a full log showing my exact steps, which following the installation instructions.

Chambana, 2023-06-20:
I believe this issue is related to CMake 3.16 the default version for Ubuntu 20.04.  

In any case, a workaround was found.  
The CMakeLists has "humble" hardcoded into a path.  Changing that, along with 2 others small tweaks, allow for this repo to compile and (as far as I've tested) work satisfactorily in Ubuntu 20.04 w/ROS2 Foxy.  

If you need this repo to work on Foxy or want to make this repo support ROS2 Foxy natively,  see my fork with changes.
https://github.com/Chambana/witmotion_IMU_ros/commit/655f745cbcf8f2e598b68127d881e479b81692f1

twdragon, 2023-06-20:
@Chambana can you prepare a PR to do that?

gsokoll, 2023-06-23:
Rather than hard coding in the ROS2 release name, I think you can address this by appropriate use of find_package() and set() in the CMakeLists.txt.  Something like:
```
# Find dependencies
find_package(ament_cmake REQUIRED)
find_package(tf2 REQUIRED)
find_package(tf2_geometry_msgs REQUIRED)
find_package(rclcpp REQUIRED)
....

set(THIS_PACKAGE_INCLUDE_DEPENDS
  tf2 
  tf2_geometry_msgs 
  rclcpp
  ...
  )

ament_target_dependencies(witmotion_ros ${THIS_PACKAGE_INCLUDE_DEPENDS})

```

tfoldi, 2023-11-10:
@gsokoll do you have a PR you can share? ran into the same issue...

tfoldi, 2023-11-10:
nevermind, I did for myself. shall I share a PR?  I did changes like instead of using `qt5_wrap_cpp` I went for `AUTOMOC`:

```
# JY901

add_library(witmotion-jy901 SHARED
    src/jy901-uart.cpp
    include/witmotion/jy901-uart.h
    )
  target_link_libraries(witmotion-jy901 witmotion-wt901 Qt5::Core)
  add_executable(witmotionctl-jy901
      src/jy901-control.cpp
    )
  target_link_libraries(witmotionctl-jy901 witmotion-jy901 Qt5::Core)

set_target_properties(witmotion-jy901 PROPERTIES AUTOMOC TRUE)
```

for the top level `CMakeLists.txt` I did what gsokoll suggested:

```

set(dependencies
  tf2
  tf2_geometry_msgs
  rclcpp
  std_srvs
  sensor_msgs
  std_msgs
  rcl_interfaces
  rclcpp_lifecycle
  lifecycle_msgs
  )
ament_target_dependencies(witmotion_ros ${dependencies})
```

it builds on my Mac as well

gsokoll, 2023-11-11:
@tfoldi a PR would be great.

tfoldi, 2023-12-08:
@gsokoll can you have a look at #36? does it look right?

twdragon, 2023-12-29:
@tfoldi elaborating #36