ElettraSciComp/witmotion_IMU_ros #32: BWT901CL imu not showing any data

shazib2t, 2023-10-05:
I was able to run the wit-motion imu roslaunch file and it seems to connect as shown below, but I do not see the /imu data in the ros topic.

`$ roslaunch witmotion_ros wt901.launch `
... logging to /home/aro/.ros/log/5d613466-638a-11ee-b39d-efb92b729d2e/roslaunch-aro-15176.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://aro:42723/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.15
 * /witmotion_imu/altimeter_publisher/addition: 0.0
 * /witmotion_imu/altimeter_publisher/coefficient: 1.0
 * /witmotion_imu/altimeter_publisher/enabled: False
 * /witmotion_imu/altimeter_publisher/topic_name: /altitude
 * /witmotion_imu/barometer_publisher/addition: 0.0
 * /witmotion_imu/barometer_publisher/coefficient: 1.0
 * /witmotion_imu/barometer_publisher/enabled: False
 * /witmotion_imu/barometer_publisher/frame_id: base_link
 * /witmotion_imu/barometer_publisher/topic_name: /barometer
 * /witmotion_imu/barometer_publisher/variance: 0.001
 * /witmotion_imu/baud_rate: 115200
 * /witmotion_imu/gps_publisher/enabled: False
 * /witmotion_imu/gps_publisher/ground_speed_topic_name: /gps_ground_speed
 * /witmotion_imu/gps_publisher/navsat_altitude_topic_name: /gps_altitude
 * /witmotion_imu/gps_publisher/navsat_fix_frame_id: world
 * /witmotion_imu/gps_publisher/navsat_fix_topic_name: /gps
 * /witmotion_imu/gps_publisher/navsat_satellites_topic_name: /gps_satellites
 * /witmotion_imu/gps_publisher/navsat_variance_topic_name: /gps_variance
 * /witmotion_imu/imu_publisher/frame_id: imu
 * /witmotion_imu/imu_publisher/measurements/acceleration/covariance: [0.0364, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/acceleration/enabled: True
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/covariance: [0.0663, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/enabled: True
 * /witmotion_imu/imu_publisher/measurements/orientation/covariance: [0.0479, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/orientation/enabled: True
 * /witmotion_imu/imu_publisher/topic_name: /imu
 * /witmotion_imu/imu_publisher/use_native_orientation: True
 * /witmotion_imu/magnetometer_publisher/addition: 0.0
 * /witmotion_imu/magnetometer_publisher/coefficient: 1e-08
 * /witmotion_imu/magnetometer_publisher/covariance: [1.87123e-07, 0, ...
 * /witmotion_imu/magnetometer_publisher/enabled: True
 * /witmotion_imu/magnetometer_publisher/frame_id: compass
 * /witmotion_imu/magnetometer_publisher/topic_name: /magnetometer
 * /witmotion_imu/orientation_publisher/enabled: True
 * /witmotion_imu/orientation_publisher/topic_name: /orientation
 * /witmotion_imu/polling_interval: 10
 * /witmotion_imu/port: ttyUSB0
 * /witmotion_imu/restart_service_name: /restart_imu
 * /witmotion_imu/rtc_publisher/enabled: False
 * /witmotion_imu/rtc_publisher/topic_name: /witmotion_clock
 * /witmotion_imu/temperature_publisher/addition: 0.0
 * /witmotion_imu/temperature_publisher/coefficient: 1.0
 * /witmotion_imu/temperature_publisher/enabled: True
 * /witmotion_imu/temperature_publisher/frame_id: base_link
 * /witmotion_imu/temperature_publisher/from_message: magnetometer
 * /witmotion_imu/temperature_publisher/topic_name: /temperature
 * /witmotion_imu/temperature_publisher/variance: 0.01829
 * /witmotion_imu/timeout_ms: 150

NODES
  /
    witmotion_imu (witmotion_ros/witmotion_ros_node)

ROS_MASTER_URI=http://localhost:11311

process[witmotion_imu-1]: started with pid [15225]
Opening device "ttyUSB0" at 115200 baud
Instantiating timer at 10 ms

** when I am trying to see the /imu topic it does not give any data. 

NEED HELP on this

twdragon, 2023-10-05:
`/witmotion_imu/imu_publisher/use_native_orientation: True`

Try to set this parameter to `false`, because if your IMU does not produce quaternions, it will not show any data with this parameter turned `true`

twdragon, 2023-10-05:
Thanks!  This is an easy case also described here https://wiki.ros.org/witmotion_ros/Troubleshooting

closed 2023-10-05 by twdragon