ElettraSciComp/witmotion_IMU_ros #37: Issue with HWT 905-485 Sensor - Timed Out Waiting for Data

melhem17, 2023-12-17:
I am currently in the process of setting it up using the ROS driver provided on your GitHub repository.
I followed the installation instructions meticulously, including the prerequisites, and successfully built the package using the following commands:

cd catkin_ws
git clone --recursive https://github.com/ElettraSciComp/witmotion_IMU_ros.git src/witmotion_ros
catkin_make

Everything seemed to work well until I attempted to launch the sensor using the command:
roslaunch witmotion_ros witmotion.launch
However, I encountered an issue with the attached error message.

[image]

I have verified the device connection and baudrate settings, but the issue persists. 
I would greatly appreciate any guidance or troubleshooting steps you can provide to help me overcome this challenge.

twdragon, 2023-12-29:
@melhem17 can you connect the device using another program?

twdragon, 2024-02-06:
Closed due to inactivity

closed 2024-02-06 by twdragon

Albert-wa, 2025-04-01:
> [@melhem17](https://github.com/melhem17) can you connect the device using another program?

the same Timed out issue with 905-485, which program should I try to find the problem?