ElettraSciComp/witmotion_IMU_ros #41: wit WT901C-TTC not detected on my raspberry pi4

ChehabiMed, 2024-03-26:
Hello, I received lately the WT901C-TTC sensor and i connected the pins to my raspberry pi4 correctly, (Rx to Tx and Tx to Rx, Vcc to 5v ) but i cant detect the sensor on the serial using `ls /dev/tty*`  i'm using ubuntu 20.4.
Did i miss something?
also i have no access to the datasheets.

twdragon, 2024-03-26:
@ChehabiMed thank you for raising the issue because the IPFS gateways should be changed. I will fix it soon. 

For the serial, if you want to use it on Raspberry, you should at first to configure pins in `raspi-config` as normal UART port, then to try to read the bytes from them using `screen` or `minicom` with connected and powered sensor

closed 2024-03-26 by twdragon

ChehabiMed, 2024-03-27:
Thank you for your help, I was able to detect the sensor on the `ttyS0` and could visualize characters using `minicon` on a baudrate of 9600.

Now i get an error with the default parameters of the wt901 when using the driver's launch.
here is the output:

`roslaunch witmotion_ros wt901.launch 
... logging to /home/ubuntu/.ros/log/2fe389bc-ec29-11ee-8009-eb857ae380be/roslaunch-ubuntu-2302.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.141.132:41195/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /witmotion_imu/altimeter_publisher/addition: 0.0
 * /witmotion_imu/altimeter_publisher/coefficient: 1.0
 * /witmotion_imu/altimeter_publisher/enabled: False
 * /witmotion_imu/altimeter_publisher/topic_name: /altitude
 * /witmotion_imu/barometer_publisher/addition: 0.0
 * /witmotion_imu/barometer_publisher/coefficient: 1.0
 * /witmotion_imu/barometer_publisher/enabled: False
 * /witmotion_imu/barometer_publisher/frame_id: base_link
 * /witmotion_imu/barometer_publisher/topic_name: /barometer
 * /witmotion_imu/barometer_publisher/variance: 0.001
 * /witmotion_imu/baud_rate: 115200
 * /witmotion_imu/gps_publisher/enabled: False
 * /witmotion_imu/gps_publisher/ground_speed_topic_name: /gps_ground_speed
 * /witmotion_imu/gps_publisher/navsat_altitude_topic_name: /gps_altitude
 * /witmotion_imu/gps_publisher/navsat_fix_frame_id: world
 * /witmotion_imu/gps_publisher/navsat_fix_topic_name: /gps
 * /witmotion_imu/gps_publisher/navsat_satellites_topic_name: /gps_satellites
 * /witmotion_imu/gps_publisher/navsat_variance_topic_name: /gps_variance
 * /witmotion_imu/imu_publisher/frame_id: imu
 * /witmotion_imu/imu_publisher/measurements/acceleration/covariance: [0.0364, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/acceleration/enabled: True
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/covariance: [0.0663, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/enabled: True
 * /witmotion_imu/imu_publisher/measurements/orientation/covariance: [0.0479, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/orientation/enabled: True
 * /witmotion_imu/imu_publisher/topic_name: /imu
 * /witmotion_imu/imu_publisher/use_native_orientation: True
 * /witmotion_imu/magnetometer_publisher/addition: 0.0
 * /witmotion_imu/magnetometer_publisher/coefficient: 1e-08
 * /witmotion_imu/magnetometer_publisher/covariance: [1.87123e-07, 0, ...
 * /witmotion_imu/magnetometer_publisher/enabled: True
 * /witmotion_imu/magnetometer_publisher/frame_id: compass
 * /witmotion_imu/magnetometer_publisher/topic_name: /magnetometer
 * /witmotion_imu/orientation_publisher/enabled: True
 * /witmotion_imu/orientation_publisher/topic_name: /orientation
 * /witmotion_imu/polling_interval: 10
 * /witmotion_imu/port: ttyS0
 * /witmotion_imu/restart_service_name: /restart_imu
 * /witmotion_imu/rtc_publisher/enabled: False
 * /witmotion_imu/rtc_publisher/topic_name: /witmotion_clock
 * /witmotion_imu/temperature_publisher/addition: 0.0
 * /witmotion_imu/temperature_publisher/coefficient: 1.0
 * /witmotion_imu/temperature_publisher/enabled: True
 * /witmotion_imu/temperature_publisher/frame_id: base_link
 * /witmotion_imu/temperature_publisher/from_message: magnetometer
 * /witmotion_imu/temperature_publisher/topic_name: /temperature
 * /witmotion_imu/temperature_publisher/variance: 0.01829
 * /witmotion_imu/timeout_ms: 150

NODES
  /
[... 67 lines cut ...]
[roslaunch][INFO] 2024-03-27 11:30:12,783: ... successfully launched [witmotion_imu-1]
[roslaunch][INFO] 2024-03-27 11:30:12,784: ... launch_nodes complete
[roslaunch.pmon][INFO] 2024-03-27 11:30:12,784: registrations completed <ProcessMonitor(ProcessMonitor-1, started daemon 281473061581280)>
[roslaunch.parent][INFO] 2024-03-27 11:30:12,784: ... roslaunch parent running, waiting for process exit
[roslaunch][INFO] 2024-03-27 11:30:12,784: spin
[roslaunch][ERROR] 2024-03-27 11:30:18,475: [witmotion_imu-1] process has died [pid 2374, exit code -11, cmd /home/ubuntu/catkin_ws/devel/lib/witmotion_ros/witmotion_ros_node __name:=witmotion_imu __log>
log file: /home/ubuntu/.ros/log/2fe389bc-ec29-11ee-8009-eb857ae380be/witmotion_imu-1*.log
[roslaunch.pmon][INFO] 2024-03-27 11:30:18,476: ProcessMonitor.unregister[witmotion_imu-1] starting
[roslaunch.pmon][INFO] 2024-03-27 11:30:18,476: ProcessMonitor.unregister[witmotion_imu-1] complete
[roslaunch][INFO] 2024-03-27 11:30:18,476: all processes on machine have died, roslaunch will exit
[roslaunch.pmon][INFO] 2024-03-27 11:30:18,577: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 281473061581280)>
[roslaunch.pmon][INFO] 2024-03-27 11:30:18,577: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 281473061581280)>: remaining procs are []
[roslaunch.pmon][INFO] 2024-03-27 11:30:18,579: ProcessMonitor exit: cleaning up data structures and signals
[roslaunch.pmon][INFO] 2024-03-27 11:30:18,580: ProcessMonitor exit: pmon has shutdown
[roslaunch][INFO] 2024-03-27 11:30:18,593: process monitor is done spinning, initiating full shutdown
[roslaunch][INFO] 2024-03-27 11:30:18,593: runner.stop()
[roslaunch][INFO] 2024-03-27 11:30:18,593: shutting down processing monitor...
[roslaunch][INFO] 2024-03-27 11:30:18,593: shutting down processing monitor <ProcessMonitor(ProcessMonitor-1, stopped daemon 281473061581280)>
[... cut ...] h.pmon][INFO] 2024-03-27 11:30:18,594: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 281473061581280)>
[roslaunch][INFO] 2024-03-27 11:30:18,594: ... shutting down processing monitor complete
[roslaunch][INFO] 2024-03-27 11:30:18,594: done
[roslaunch.pmon][INFO] 2024-03-27 11:30:18,594: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 281473061581280)>
[rospy.core][INFO] 2024-03-27 11:30:18,594: signal_shutdown [atexit]
`
i appreciate your help.

ChehabiMed, 2024-03-27:
update:
After setting some parameters to default values and using frequency of 100Hz using this commands:
```rosrun witmotion_ros witmotionctl-wt901 -d /dev/ttyS0 --set-baudrate 9600
rosrun witmotion_ros witmotionctl-wt901 -d /dev/ttyS0 -b 115200 --set-frequency 100
rosrun witmotion_ros witmotionctl-wt901 -d /dev/ttyS0 -b 115200 -i 50
```
and this command to test the communication:
`rosrun witmotion_ros message-enumerator -d ttyS0 -b 9600 -p 50`

I get this output:

```Press Ctrl+C to stop enumeration and see the report
Opening device "ttyS0" at 9600 baud
Instantiating timer at 50 ms
ERROR: Timed out waiting for data, please check device connection and baudrate!
Closing TTL connection

WITMOTION UART MESSAGE ENUMERATOR BY TWDRAGON

Acquired at Wednesday, 27 March 2024 13:28:48 UTC

ID	Qty	Description

0x51	11	Accelerations
0x52	10	Angular velocities
0x53	10	Spatial orientation (Euler angles)
0x54	10	Magnetometer/Magnetic orientation

	Unknown IDs: 0 [  ] 
Total messages: 41
```

ChehabiMed, 2024-03-27:
Update: 
My node is publishing data now with a baud rate of 9600 and a polling time of 5ms, the only thing i did is to reboot my Rpy! apparently some changes took effect only after a reboot

twdragon, 2024-03-27:
@ChehabiMed please make sure that the hardware is able to publish on this frequency with the baudrate 9600 baud. I think, then we can close the issue if the error does not persist.

ChehabiMed, 2024-03-27:
I am running into the same problems again once i reboot my Rpy, maybe because i misunderstood the use of the commands i used!
The problem i am having now is with the port ttyS0 who disable the permissions for the groups everytime i run the launch or one of the commands i used before, is that due to communication failure?
Can you please confirm the role of the following commands:

'rosrun witmotion_ros witmotionctl-wt901 -d /dev/ttyS0 --set-baudrate 9600' to set the baudrate of the sensor to 9600
'rosrun witmotion_ros witmotionctl-wt901 -d /dev/ttyS0 -b 115200 --set-frequency 100' to set the frequency of the sensor to 100Hz
'rosrun witmotion_ros witmotionctl-wt901 -d /dev/ttyS0 -b 115200 -i 50' to set the polling time on the sensor to 50ms

I did notice that i am required to reconnect after each command, do that means to reconnect the cables of the sensor or reboot the system?

Thank you for your clarifications

commit 30538ed0 (2024-03-26): Update README.md

Fixes IPFS link. Closes #41