ElettraSciComp/witmotion_IMU_ros #42: [ROS2] Can't read anything on `/imu` topic

YuilTripathee, 2024-06-26:
I feel like I'm missing something. I'm using multiple sensors (JY901, WT901C, WT901BLE version). All of them read perfectly with standard Witmotion SDK and `witmotion-debug` tool from `pip`. But, when I plug in those MCU and launch `ros2` node, it takes the USB serial connection and no data seem to be published on `\imu`. However, a topic is registered and there seems to be 1 publisher too.

Normally, on Pixhawk 4 and Mavros this was spontaneous.

Do I need to activate something within the node or issue is something else?

twdragon, 2024-06-26:
Can you publish your parameters list and logfile?

YuilTripathee, 2024-06-26:
I simply launched it with the launch (py) file.

```bash
rockie@rockie-HP-Laptop-15s-fq1xxx:~/THEATERS/GARAGE/G3-ACTIVITIES/UAC1-Robotics_RnD/repos/TRCC_Sketches_22_26$ ros2 launch witmotion_ros wt901_launch.py
[INFO] [launch]: All log files can be found below /home/rockie/.ros/log/2024-06-26-16-27-24-260591-rockie-HP-Laptop-15s-fq1xxx-65686
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [witmotion_ros_node-1]: process started with pid [65687]
[witmotion_ros_node-1] [INFO] [1719394044.408199134] [witmotion]: Starting node with lib version (1.2.28~dev_5c2e86d).
[witmotion_ros_node-1] [INFO] [1719394044.408490491] [ROSWitmotionSensorController]: Controller started
[witmotion_ros_node-1] [INFO] [1719394044.408649030] [MinimalPublisher]: QT spin !!!!!
```

Here's the recent log:
```bash
1719394044.2622082 [INFO] [launch]: All log files can be found below /home/rockie/.ros/log/2024-06-26-16-27-24-260591-rockie-HP-Laptop-15s-fq1xxx-65686
1719394044.2624881 [INFO] [launch]: Default logging verbosity is set to INFO
1719394044.3624544 [INFO] [witmotion_ros_node-1]: process started with pid [65687]
1719394044.4087701 [witmotion_ros_node-1] [INFO] [1719394044.408199134] [witmotion]: Starting node with lib version (1.2.28~dev_5c2e86d).
1719394044.4096971 [witmotion_ros_node-1] [INFO] [1719394044.408490491] [ROSWitmotionSensorController]: Controller started
1719394044.4099264 [witmotion_ros_node-1] [INFO] [1719394044.408649030] [MinimalPublisher]: QT spin !!!!!
1719394044.4104037 [witmotion_ros_node-1] Opening device "ttyUSB0" at 115200 baud
1719394044.4108572 [witmotion_ros_node-1] Instantiating timer at 10 ms
```

twdragon, 2024-06-26:
@YuilTripathee you need to inspect thoroughly the parameters. If there is any declared datapacket that is not actually produced by the IMU, the node will never construct the message to publish

YuilTripathee, 2024-06-26:
@twdragon Thank you very much. Now, it works for Yahboom 10 axis IMU (except for barometer and altitude) and WT901C-232. I can't read `\orientation` topic but I can get the quarternions from `\imu` topic itself.

Here's my new setup, I'd be happy to put a PR for config of this new IMU device.

```yml
witmotion:
  ros__parameters:
    port: ttyUSB0
    baud_rate: 9600 # baud
    polling_interval: 50 # ms
    timeout_ms: 150 # ms
    restart_service_name: /restart_imu
    imu_publisher:
      topic_name: /imu
      frame_id: imu
      use_native_orientation: false # 311 edit: default -> true (ref: https://github.com/ElettraSciComp/witmotion_IMU_ros/issues/34)
      measurements:
        acceleration:
          enabled: true
          covariance: [0.0364, 0.0, 0.0, 0.0, 0.0048, 0.0, 0.0, 0.0, 0.0796]
        angular_velocity:
          enabled: true
          covariance: [0.0663, 0.0, 0.0, 0.0, 0.1453, 0.0, 0.0, 0.0, 0.0378]
        orientation:
          enabled: true
          covariance: [0.0479, 0.0, 0.0, 0.0, 0.0207, 0.0, 0.0, 0.0, 0.0041]
    temperature_publisher:
      enabled: true # 311: change from true to false
      topic_name: /temperature
      frame_id: base_link
      from_message: acceleration # acceleration, angular_vel, orientation, magnetometer
      variance: 0.01829
      coefficient: 1.0 # Linear calibration parameters: coefficient
      addition: 0.0 # and addendum
    magnetometer_publisher:
      enabled: true
      topic_name: /magnetometer
      frame_id: compass
      coefficient: 0.00000001 # Linear calibration parameters: coefficient
      addition: 0.0 # and addendum
      covariance:
        [
          0.000000187123,
          0.0,
          0.0,
          0.0,
          0.000000105373,
          0.0,
          0.0,
          0.0,
          0.000000165816,
        ]
    barometer_publisher:
      enabled: false
      topic_name: /barometer
      frame_id: base_link
      variance: 0.001
      coefficient: 1.0 # Linear calibration parameters: coefficient
      addition: 0.0 # and addendum
    altimeter_publisher:
      enabled: false
      topic_name: /altitude
      coefficient: 1.0 # Linear calibration parameters: coefficient
      addition: 0.0 # and addendum
    orientation_publisher:
      enabled: false
      topic_name: /orientation
    gps_publisher:
      enabled: false
      navsat_fix_frame_id: world
      navsat_fix_topic_name: /gps
      navsat_altitude_topic_name: /gps_altitude
      navsat_satellites_topic_name: /gps_satellites
      navsat_variance_topic_name: /gps_variance
      ground_speed_topic_name: /gps_ground_speed
    rtc_publisher:
      enabled: false
      topic_name: /witmotion_clock
```

closed 2025-04-01 by twdragon