ElettraSciComp/witmotion_IMU_ros #44: Can't receive any message from WT901BLECL5.0

pcy028x2, 2024-10-25:
PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /witmotion_imu/altimeter_publisher/addition: 0.0
 * /witmotion_imu/altimeter_publisher/coefficient: 1.0
 * /witmotion_imu/altimeter_publisher/enabled: True
 * /witmotion_imu/altimeter_publisher/topic_name: /altitude
 * /witmotion_imu/barometer_publisher/addition: 0.0
 * /witmotion_imu/barometer_publisher/coefficient: 1.0
 * /witmotion_imu/barometer_publisher/enabled: True
 * /witmotion_imu/barometer_publisher/frame_id: base_link
 * /witmotion_imu/barometer_publisher/topic_name: /barometer
 * /witmotion_imu/barometer_publisher/variance: 0.001
 * /witmotion_imu/baud_rate: 115200
 * /witmotion_imu/gps_publisher/enabled: False
 * /witmotion_imu/gps_publisher/ground_speed_topic_name: /gps_ground_speed
 * /witmotion_imu/gps_publisher/navsat_altitude_topic_name: /gps_altitude
 * /witmotion_imu/gps_publisher/navsat_fix_frame_id: world
 * /witmotion_imu/gps_publisher/navsat_fix_topic_name: /gps
 * /witmotion_imu/gps_publisher/navsat_satellites_topic_name: /gps_satellites
 * /witmotion_imu/gps_publisher/navsat_variance_topic_name: /gps_variance
 * /witmotion_imu/imu_publisher/frame_id: imu
 * /witmotion_imu/imu_publisher/measurements/acceleration/covariance: [0.0088, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/acceleration/enabled: True
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/covariance: [0.1199, 0, 0, 0,...
 * /witmotion_imu/imu_publisher/measurements/angular_velocity/enabled: True
 * /witmotion_imu/imu_publisher/measurements/orientation/covariance: [0.019, 0, 0, 0, ...
 * /witmotion_imu/imu_publisher/measurements/orientation/enabled: True
 * /witmotion_imu/imu_publisher/topic_name: /imu
 * /witmotion_imu/imu_publisher/use_native_orientation: False
 * /witmotion_imu/magnetometer_publisher/addition: 0.0
 * /witmotion_imu/magnetometer_publisher/coefficient: 1e-08
 * /witmotion_imu/magnetometer_publisher/covariance: [1.87123e-07, 0, ...
 * /witmotion_imu/magnetometer_publisher/enabled: True
 * /witmotion_imu/magnetometer_publisher/frame_id: imu
 * /witmotion_imu/magnetometer_publisher/topic_name: /magnetometer
 * /witmotion_imu/orientation_publisher/enabled: True
 * /witmotion_imu/orientation_publisher/topic_name: /orientation
 * /witmotion_imu/polling_interval: 50
 * /witmotion_imu/port: ttyUSB0
 * /witmotion_imu/restart_service_name: /restart_imu
 * /witmotion_imu/rtc_publisher/enabled: True
 * /witmotion_imu/rtc_publisher/presync: True
 * /witmotion_imu/rtc_publisher/topic_name: /witmotion_clock
 * /witmotion_imu/temperature_publisher/addition: 0.0
 * /witmotion_imu/temperature_publisher/coefficient: 1.0
 * /witmotion_imu/temperature_publisher/enabled: True
 * /witmotion_imu/temperature_publisher/frame_id: base_link
 * /witmotion_imu/temperature_publisher/from_message: magnetometer
 * /witmotion_imu/temperature_publisher/topic_name: /temperature
 * /witmotion_imu/temperature_publisher/variance: 0.02683
 * /witmotion_imu/timeout_ms: 150

NODES
  /
    witmotion_imu (witmotion_ros/witmotion_ros_node)

This is my parameters, I could see the topics (/imu) are generated, but when i tried to see the messages using `rostopic echo /imu`, it does not work.
Using `rostopic hz /imu`, it shows `no new messages`.
I changed my baudrate 115200 to 9600 but the trouble still remains.

twdragon, 2024-10-25:
@pcy028x2 did you try with `message_enumerator`? Which Witmotion IMU messages it shows available on the stream?

pcy028x2, 2024-10-29:
> @pcy028x2 did you try with `message_enumerator`? Which Witmotion IMU messages are available on the stream? Consider also removing RTC clock messages as the incorrect publisher's request may prevent the IMU from generating a proper stream.

Sorry for giving a late response, yes I did but I couldn't find nothing using `message_enumerator`. Instead I coped with this problem by using https://github.com/WITMOTION/WitBluetooth_BWT901BLE5_0 this repository. Thanks for your fast reply.

closed 2024-10-29 by pcy028x2