ElettraSciComp/witmotion_IMU_ros #45: HWT906 not showing any data

musstt, 2024-11-06:
this is my config.yaml file 
witmotion_imu:
    port: ttyUSB0
    baud_rate: 115200 # baud
    polling_interval: 50 # ms
    timeout_ms: 150 # ms
    restart_service_name: /restart_imu
    imu_publisher:
        topic_name: /imu
        frame_id: imu
        use_native_orientation: false
        measurements:
            acceleration:
                enabled: true
                covariance: [ 0.0088,    0,       0,
                              0,         0.0550,  0,
                              0,         0,       0.0267 ]
            angular_velocity:
                enabled: true
                covariance: [ 0.1199,    0,       0,
                              0,         0.5753,  0,
                              0,         0,       0.0267 ]
            orientation:
                enabled: true
                covariance: [ 0.0190,    0,       0,
                              0,         0.0120,  0,
                              0,         0,       0.0107 ]
    temperature_publisher:
        enabled: false
        topic_name: /temperature
        frame_id: base_link
        from_message: magnetometer # acceleration, angular_vel, orientation, magnetometer
        variance: 0.02683
        coefficient: 1.0 # Linear calibration parameters: coefficient
        addition: 0.0 # and addendum
    magnetometer_publisher:
        enabled: true
        topic_name: /magnetometer
        frame_id: imu
        coefficient: 0.00000001 # Linear calibration parameters: coefficient
        addition: 0.0 # and addendum
        covariance: [ 0.000000187123,   0,              0,
                      0,                0.000000105373, 0,
                      0,                0,              0.000000165816  ]
    barometer_publisher:
        enabled: true
        topic_name: /barometer
        frame_id: base_link
        variance: 0.001
        coefficient: 1.0 # Linear calibration parameters: coefficient
        addition: 0.0 # and addendum
    altimeter_publisher:
        enabled: true
        topic_name: /altitude
        coefficient: 1.0 # Linear calibration parameters: coefficient
        addition: 0.0 # and addendum
    orientation_publisher:
        enabled: true
        topic_name: /orientation
    gps_publisher:
        enabled: false
        navsat_fix_frame_id: world
        navsat_fix_topic_name: /gps
        navsat_altitude_topic_name: /gps_altitude
        navsat_satellites_topic_name: /gps_satellites
        navsat_variance_topic_name: /gps_variance
        ground_speed_topic_name: /gps_ground_speed
    rtc_publisher:
        enabled: true
        topic_name: /witmotion_clock
        presync: true

i dont know what parameter need to be set true and false

twdragon, 2024-11-06:
@musstt you should disable anything except Accelerations in the IMU ROS message builder, otherwise it will never consider the data structure complete, so:
```diff
    imu_publisher:
        topic_name: /imu
        frame_id: imu
        use_native_orientation: false
        measurements:
            acceleration:
                enabled: true
                covariance: [ 0.0088,    0,       0,
                              0,         0.0550,  0,
                              0,         0,       0.0267 ]
            angular_velocity:
-                 enabled: true
+                enabled: false
                covariance: [ 0.1199,    0,       0,
                              0,         0.5753,  0,
                              0,         0,       0.0267 ]
            orientation:
-                 enabled: true
+                enabled: false
                covariance: [ 0.0190,    0,       0,
                              0,         0.0120,  0,
                              0,         0,       0.0107 ]
```

twdragon, 2024-11-06:
@musstt if you will have success, please do not hesitate to create the test report here: #16

musstt, 2024-11-06:
what about the baud rate and polling interval parameters?

musstt, 2024-11-06:
and another publisher? should i change the enable parameters to false?

musstt, 2024-11-06:
[image]
what do you think?

musstt, 2024-11-06:
[image]

twdragon, 2024-11-06:
@musstt I recommend you reduce the baudrate to 115200, adjust the polling rate to 10 ms, and output rate to 100 Hz, and then set up the bandwidth accordingly. 1 KHz, as I see, will be less accurate (due to the Kalman filter) and redundant for most applications.

twdragon, 2024-11-06:
@musstt sorry but I need to close this issue here, as your device is working and there are no bugs reported. Please, if you need still to reach me, do not hesitate to open a discussion on [ROS Discourse](https://discourse.ros.org/) mentioning me as `@twdragon` and inserting a link to this issue (`#45`)

closed 2024-11-06 by twdragon

musstt, 2024-11-06:
Wait, could you tell me what parameters enable should be true or false, i cant figure it out although i see from the windows application, sorry

musstt, 2024-11-06:
Thank you so much!!!!! It works!!!!!