ElettraSciComp/witmotion_IMU_ros #47: Sensor error: Timed out waiting for data, please check device connection and baudrate!

Vikas-106, 2025-03-10:
i am using wit motion imu model : WT901BLECL5.0 with ros1 noetic . when running this package after changing the port in yaml file got this error .
"""
Configuration task detected, 1 commands in list, configuring sensor...
Sending configuration packet 0x0
Configuration packet sent, flushing buffers...
Configuration completed
[ INFO] [1741616291.501104001]: RTC synchronized
[ERROR] [1741616291.501556259]: Sensor error: Timed out waiting for data, please check device connection and baudrate!
[ INFO] [1741616291.501636634]: Entering SUSPENDED state
QObject::killTimer: Timers cannot be stopped from another thread
QObject::~QObject: Timers cannot be stopped from another thread
QSocketNotifier: Socket notifiers cannot be enabled or disabled from another thread
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1741616291.502397837]: Sensor error: Timed out waiting for data, please check device connection and baudrate!
[ INFO] [1741616291.502431750]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1741616291.502474800]: Sensor error: Timed out waiting for data, please check device connection and baudrate!
[ INFO] [1741616291.502500497]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!
[ERROR] [1741616291.502537207]: Sensor error: Timed out waiting for data, please check device connection and baudrate!
[ INFO] [1741616291.502562271]: Entering SUSPENDED state
Suspending TTL connection, please emit RunPoll() again to proceed!
[witmotion_imu-1] process has died [pid 25186, exit code -11, cmd /home/vikas/Restaurant_ws/devel/lib/witmotion_ros/witmotion_ros_node __name:=witmotion_imu __log:=/home/vikas/.ros/log/9eee1fb4-fdb9-11ef-a9d9-b3401817fed5/witmotion_imu-1.log].
log file: /home/vikas/.ros/log/9eee1fb4-fdb9-11ef-a9d9-b3401817fed5/witmotion_imu-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
""""
what can i do ?

Prabuddha781, 2025-03-13:
same problem with me as well and same model:

```
pierre@pierre:~/imu_ws$ ros2 launch witmotion_ros wt901_launch.py
[INFO] [launch]: All log files can be found below /home/pierre/.ros/log/2025-03-12-19-50-56-195582-pierre-28650
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [witmotion_ros_node-1]: process started with pid [28651]
[witmotion_ros_node-1] [INFO] [1741834256.262997499] [witmotion]: Starting node with lib version (1.2.28~dev_5c2e86d).
[witmotion_ros_node-1] [INFO] [1741834256.263123086] [ROSWitmotionSensorController]: Controller started
[witmotion_ros_node-1] [INFO] [1741834256.263194687] [MinimalPublisher]: QT spin !!!!!
[witmotion_ros_node-1] [ERROR] [1741834261.265109704] [ROSWitmotionSensorController]: Sensor error: Timed out waiting for data, please check device connection and baudrate!
[witmotion_ros_node-1] [INFO] [1741834261.265160129] [ROSWitmotionSensorController]: Entering SUSPENDED state
[witmotion_ros_node-1] QObject::killTimer: Timers cannot be stopped from another thread
[witmotion_ros_node-1] QObject::~QObject: Timers cannot be stopped from another thread
[witmotion_ros_node-1] QSocketNotifier: Socket notifiers cannot be enabled or disabled from another thread
[ERROR] [witmotion_ros_node-1]: process has died [pid 28651, exit code -11, cmd '/home/pierre/imu_ws/install/witmotion_ros/lib/witmotion_ros/witmotion_ros_node --ros-args --params-file /home/pierre/imu_ws/install/witmotion_ros/share/witmotion_ros/config/wt901.yml'].

```

Prabuddha781, 2025-03-13:
@Vikas-106 did you find a solution yet?

twdragon, 2025-03-13:
Please use `message-enumerator` to see which messages the device really sends

Vikas-106, 2025-03-17:
i used "screen /dev/ttyACM0 115200" to check if the data is received from the port , yes i see the data but it shows random symbol , i changed the baud rate but still getting the random symbol . i tried all baud rate from the manual still no result . Also when running 
"roslaunch witmotion_ros wt901.launch " i get 
""""
 * /witmotion_imu/temperature_publisher/addition: 0.0
 * /witmotion_imu/temperature_publisher/coefficient: 1.0
 * /witmotion_imu/temperature_publisher/enabled: True
 * /witmotion_imu/temperature_publisher/frame_id: base_link
 * /witmotion_imu/temperature_publisher/from_message: magnetometer
 * /witmotion_imu/temperature_publisher/topic_name: /temperature
 * /witmotion_imu/temperature_publisher/variance: 0.01829
 * /witmotion_imu/timeout_ms: 2000

NODES
  /
    witmotion_imu (witmotion_ros/witmotion_ros_node)

auto-starting new master
process[master]: started with pid [9786]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 90d66730-0325-11f0-909f-31f916258e6e
process[rosout-1]: started with pid [9808]
started core service [/rosout]
process[witmotion_imu-2]: started with pid [9815]
Opening device "ttyACM0" at 9600 baud
Instantiating timer at 10 ms
"""
vikas@vikas:~$ rostopic list 
/imu
/magnetometer
/orientation
/rosout
/rosout_agg
/temperature
vikas@vikas:~$ rostopic echo /imu 
"""
topic is visible but no data is published . don't know what to do

twdragon, 2025-03-17:
@Vikas-106 come on! It was developed on Linux and should be compiled on every operating system supporting Qt. Try to recompile the Witmotion package and use it from `rosrun`

Vikas-106, 2025-03-17:
ah sorry bro didn't source it , 
vikas@vikas:~/Restaurant_ws$ rosrun witmotion_ros message-enumerator 
Press Ctrl+C to stop enumeration and see the report
Opening device "ttyACM0" at 9600 baud
Instantiating timer at 50 ms
^C
Closing TTL connection

WITMOTION UART MESSAGE ENUMERATOR BY TWDRAGON

Acquired at Monday, 17 March, 2025 5:33:35 PM IST

ID	Qty	Description

	Unknown IDs: 0 [  ] 
Total messages: 0

this is what i am getting

twdragon, 2025-03-17:
@Vikas-106 try to experiment with port parameters to see on which mode you receive the proper data the driver is able to decode

Vikas-106, 2025-03-17:
which port parameters bro , the thing in config file or about the permissions to the port in my system . "I am completely new to this so sorry if i ask some silly questions "

twdragon, 2025-03-17:
@Vikas-106 the serial port, the device name and baudrate

twdragon, 2025-04-01:
Closed as inactive

closed 2025-04-01 by twdragon